robotology/yarp #341: http connections broken without ACE

paulfitz, 2015-01-05:
Commit b2ef02864ad1bef13e6a643fdcb56592b84e0731 appears to have broken http connections for yarp compiled without ACE.  To replicate:

 * Compile yarp without ACE.
 * Start the name server.
 * Follow the link mentioned on "Name server can be browsed at" line.
 * Result is blank (should be a webpage, compare with result when compiled with ACE).

paulfitz, 2015-01-05:
Result is blank in chrome, or "The connection to the server was reset while the page was loading." in firefox.

drdanz, 2015-01-07:
Actually on my computer I can see the webpage for a second before the connection is reset, and using wget I get the html, but it hangs.

It looks like that the cause is the ``l_linger`` value of the ``linger`` struct that is set to ``0``.

```c++
    struct linger lval;
    lval.l_onoff = 1;
    lval.l_linger = 0;
    stream.set_option (SOL_SOCKET, SO_LINGER, &lval, sizeof(linger));
```
Changing it to any 

According to the man page

> SO_LINGER
> Sets or gets the SO_LINGER option.  The argument is a linger structure.
>
>```c
>      struct linger {
>          int l_onoff;    /* linger active */
>          int l_linger;   /* how many seconds to linger for */
>      };
>```
>
> When  enabled, a close(2) or shutdown(2) will not return until all queued messages for the socket have been successfully sent or the linger timeout has been reached.  Otherwise, the call returns immediately and the closing is done in the background.  When the socket is closed as part of exit(2), it always lingers in the background.

Setting l_onoff to ``1`` and l_linger to ``0`` causes the connection to be aborted and any pending data is immediately discarded when ``close()`` or ``shutdown()`` are called, and therefore the webpage is not displayed by the browser. I'm not sure of the reason why this works for ACE and does not work without ACE, I was expecting it to be a pure wrapper of the system call. Also I'm not sure of the reason why we enable it...
I think we can either remove the ``SO_LINGER`` option for SKIP_ACE (or for both cases) or change the ``l_linger`` value to a non-zero value (either for SKIP_ACE only or for both cases). @paulfitz opinion?

paulfitz, 2015-01-07:
Also unsure what ACE is doing that is different. I think the current linger setup looks ok for regular yarp tcp protocol, when streaming data. In that case there is a higher-level ack so yarp will wait for data to go through before closing under normal situations, and it is not good in practice to wait around in error situations (sockets can pile up quickly, and things aren't going to get better). But it looks dodgy for RPC stuff and carriers with no acks available to yarp. Will poke at this some more. Thanks for looking at it @drdanz.

paulfitz, 2015-01-22:
I'm not getting the time to look at this carefully. Is it ok with you if I go ahead and remove the `SO_LINGER` option for SKIP_ACE @drdanz, while leaving everything else unchanged? That will make http usable under ACE again, which is a better situation than the status quo. Still need to figure this stuff out for real, but I feel bad leaving http broken.

drdanz, 2015-01-22:
Yes, sure... Perhaps leave a comment so that I don't see it in 2 months and add it again :)

paulfitz, 2015-01-23:
huh! Just noticed that the command allegedly setting linger when ACE is active is:
```
stream.set_option (ACE_IPPROTO_TCP, SO_LINGER, ...)
```
but that should be `SOL_SOCKET` not `IPPROTO_TCP` - I'm guessing now this line has just always been ineffective.

paulfitz, 2015-01-23:
Confirmed that fixing the lines active when ACE is present to `stream.set_option(SOL_SOCKET, SO_LINGER, ...)` breaks http for that configuration too.

closed 2015-01-26 by paulfitz