robotology/yarp #346: YARP_CONFIGURE_EXTERNAL_INSTALLATION : if (YARP_INSTALL_PREFIX)

jgvictores, 2015-01-11:
I'm seeing that a section of the YARP_CONFIGURE_EXTERNAL_INSTALLATION is being skipped on my (Ubuntu) machine, specifically the contents of the 'if (YARP_INSTALL_PREFIX) #if YARP is installed'.
My YARPConfig.cmake contains 'set(YARP_INSTALL_PREFIX "")' , so the check evaluates to false (complying with http://www.cmake.org/cmake/help/v3.0/command/if.html for an empty string).

1) Should 'if (YARP_INSTALL_PREFIX)' be replaced with 'if (YARP_FOUND)'?
2) Is there something wrong with my YARPConfig.cmake's 'YARP_INSTALL_PREFIX'?

paulfitz, 2015-01-12:
If you are in a yarp build directory, you'll see a YARPConfig.cmake with `set(YARP_INSTALL_PREFIX "")`, and also a YARPConfigForInstall.cmake that is similar but has a valid YARP_INSTALL_PREFIX. That second one would be copied to an *installed* YARPConfig.cmake file when yarp is installed.  So the test in YarpInstallationHelpers seems to match the comment for it, it will pass if the version of yarp being used has been installed.

I'm guessing that in your case yarp has not been installed per se, just built. I'm also guessing that that's the way you want it, but that there's some stuff that is being skipped that would be useful to you none the less. Can you talk more about the problem the skipped code causes you, and what you'd like to see happen?

@drdanz can probably be way more helpful though...

jgvictores, 2015-01-12:
@paulfitz The files (build/YARPConfig.cmake, build/YARPConfigForInstall.cmake, and installed YARPConfig.cmake) are all correctly as you described. From what I understand, what was happening is that a reminiscent $YARP_DIR environment variable was shadowing the installed YARPConfig.cmake during the CMaking.

Is it safe to drop the $YARP_ROOT and $YARP_DIR environment variables (as a general statement for Linux and Windows, at least in the sense of manually setting them) and close this issue?

paulfitz, 2015-01-12:
I personally don't have YARP_ROOT or YARP_DIR set in my environment, for what it is worth.

`YARP_DIR`, if present, will have an effect on cmake, which is sensitive to any environment variable of the form `<package>_DIR` when searching for `<package>` with `FIND_PACKAGE`. `YARP_DIR` has no special role any more apart from this use to direct cmake. If you don't want that effect, removing it sounds like the right thing to do for your setup.

`YARP_ROOT` no longer plays a role in building yarp. I doubt if the iCub project depends on it any more either.  There are occasional references to it in old examples and scripts.

jgvictores, 2015-01-12:
Okay, that works for me. And I'm happy with fewer environment variables anyway. :-)

closed 2015-01-12 by jgvictores