robotology/yarp #350: ROS2 compatibility

paulfitz, 2015-01-14:
ROS2 isn't really a thing yet, but it'll be good to be ready for it.  This is an issue to track progress. General plan is:

 * Track the ROS2 examples being maintained here https://github.com/ros2/examples
 * Check what parts of ros2 we can usefully disentangle from build system and link to from a carrier
 * Absorb [developer documentation](https://github.com/ros2/ros_core_documentation/blob/master/doc/source/index.rst)

Instructions for compiling examples are at https://github.com/ros2/examples/wiki. For testing, installing PrismTech OpenSplice from OSRF deb packages seems the least annoying of the DDS options.

traversaro, 2021-05-12:
Related repo: https://github.com/robotology-playground/yarp-ros2 .

randaz81, 2025-08-20:
The repo mentioned above was renamed as: https://github.com/robotology/yarp-devices-ros2