robotology/yarp #353: setEnvelope is the bottleneck of ControlBoardWrapper::run()

traversaro, 2015-01-15:
Doing a quick profiling of a fakebot standard run with [callgrind](http://valgrind.org/docs/manual/cl-manual.html) we discovered that (apart from a spurious string allocation that can be easily removed:
https://github.com/robotology/yarp/blob/2732dc088adf1be2edb6300e64fb4cfded505f7d/src/libYARP_dev/src/modules/ControlBoardWrapper/ControlBoardWrapper.cpp#L476 ) the main bottleneck is the setEnvelope call, that is dominating the time of the run(). 

Roughly speaking, by commenting the setEnvelope calls the mean duration of the run() on my machine drops from `0.14` ms to `0.04` ms. This may not seem a lot, but on ancient hardware (like the pc104) with a lot of wrappers running at high frequency it can be relevant. 
 
Inspecting the setEnvelope implementation it appears that this bottleneck is due to a bunch of temporary buffers that are allocated at each setEnvelope call. I could have some idea on how to remove this allocations, but given that they are deep inside  the "sancta sanctorum" of YARP I guess that @paulfitz can provide his point of view on the issue.

traversaro, 2015-01-15:
Example of callgrind dump where setEnvelope dominates `ControlBoardWrapper::run()`
[callgrind.out.17241 | uploaded via ZenHub](https://files.zenhub.io/54b7838ca249c8781446ceef)

traversaro, 2015-01-15:
Similar results are obtained when using callgrind on the robotInterface running just the motor control on the left_leg (with no readers connected to the ports).

[callgrind.out.19580 | uploaded via ZenHub](https://files.zenhub.io/54b7c93fa249c8781446d069)

Note: you can read this dump using kcachegrind .

paulfitz, 2015-01-15:
yes I'd expect plenty of room for improvement in the envelope implementation, will look at it. Btw please don't think of the yarp internals as a "sancta sanctorum", we need more people stomping through there in muddy boots tapping on things with hammers.

traversaro, 2015-01-15:
Ok understood no "sancta sanctorum"..  :) 
One possible idea that could be useful also in other part of the codebase would be to have a variant of the `BufferedConnectionWriter`class optimized for reducing memory allocation, declared as a attribute of the PortSomething class.

traversaro, 2015-01-18:
First try on wearing some "muddy boots". 

Move a `BufferedConnectionWriter` object to be an attribute would not be sufficient. The internal memory buffer is deleted (or lost, I don't understand how it is possible that we don't have memory leaks related to the `pool` attribute) at each call of the `reset()` method. For reusing the same buffer without further memory (re)allocation, it is necessary to modify the `reset()` method (or add a new one) that marks all the bytes of the buffer as unused, without deleting it. 

After this is done, the tricky part would be perform the string conversion in place, without any temporary string allocation. This should be possible using using an internal string buffer, and adding a method that returns a reference to the internal string instead of returning a string copy (as `toString()` does). `yarp::os::ConstString` is not exposing the `reserve` method that would be helpful for that, but I guess that a couple of `resize` without any `shrink_to_fit` would do the same job of `reserve` on most implementations.

paulfitz, 2015-01-19:
Yay muddy boots!

Definitely should add `reserve`.

I've working on a version of the `BufferedConnectionWriter` that reuses everything on the case of transmitting a message of the same structure. YARP1 was like this, and I promised myself to do the same in YARP2 the first time anyone noticed :-).  Also trying a few other optimizations, none of which have had a significant impact yet.

For string stuff, there are a few issues to add to what you raise. 

One annoying thing about the string class in yarp is that it is relatively inefficient due to a series of unfortunate events:

 * We want to use `std::string`, but
 * Using `std::string` in a header file associated with a dll on Windows causes sadness.
 * So we use a wrapped version of `std::string`, causing extra memory allocation.

On Linux + OSX it is perfectly fine to replace `ConstString` everywhere with `std::string`, there is a flag for that (set `YARP_WRAP_STL_STRING` to false and you should get ConstString being exactly a typedef of std::string). All that said, this particular issue doesn't seem too relevant to you from some quick tests.

Another issue is that there's no real need in the first place to convert the timestamp to a string.  That is there only because there was nowhere for timestamps in the old crufty yarp protocol passed down from the dawn of time, so I snuck it into a slot that unfortunately had to be text.  It would be easy to extend the protocol. It is goofy to be fiddling around with serializing a float to text when we can just ship the raw bytes.

traversaro, 2015-01-19:
Yes, getting rid of string conversion would be ideal, we discussed that with @lornat75 but he was afraid for backward compatibility.

traversaro, 2015-01-19:
By the way, two other things that I locally changed to optimize `ControlBoardWrapper::run()`, worth to try if you are doing some profiling:
* Remove the unused temporary string in https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/modules/ControlBoardWrapper/ControlBoardWrapper.cpp#L476
* Modify the `YARP_DEBUG` macro definition to properly disappear in Release mode, currently nothing is printed in release mode but temporary strings contained in YARP_DEBUG (for example resulting from concatenations) are still evaluated.

lornat75, 2015-01-19:
+1 for adding proper slot for timestamp and frame number.
+1 for cleaning up ControlBoardWrapper

traversaro, 2015-05-13:
cc @barbalberto that is working on reducing robotInterface load on walkman. In particular the test infrastructure to spot unwanted dynamic memory allocation introduced in https://github.com/robotology/yarp/commit/9ba5daad47653ad42d0560dfd44c99d50e309193 can be useful to debug the wrappers performance. 
Ideally no dynamic memory allocation/deallocation should happen in the wrappers run(), apart for the first run where it may be necessary to allocate the buffers, and it is relatively easy to write a test for this.

traversaro, 2015-05-21:
cc @iron76 this is one issue that maybe critical for running `wholeBodyDynamicsTree` on the PC104 .

traversaro, 2017-11-20:
We should check if this is solved by https://github.com/robotology/yarp/pull/1454 .