robotology/yarp #354: icub-main cannot find resources on Windows:_installation_from_sources (tested on VS10)

jgvictores, 2015-01-15:
From what I'm seeing, it's due to the insertion of an extra whitespace before appending "contexts", resulting in a failed: "%ICUB_DIR%/share/iCub /contexts" rather than a functional "%ICUB_DIR%/share/iCub/contexts". Therefore, for example, "contexts/simConfig/simulator.ini" is not found for iCUB_SIM.

PS: Related with https://github.com/robotology/yarp/issues/336 discussion, but separating the issue as it's not a shadowing issue.

paulfitz, 2015-01-15:
@jgvictores could you do me a favor and paste the output of `yarp resource --find foo.ini --context bar`, just to nail down what it is doing?

jgvictores, 2015-01-15:
Tried now and icub_SIM now DOES find the resources... haven't reset the machine, maybe a small session change from remote to physical desktop, or some time windows needed to refresh its environment variables...
Really sorry to bug about that, but happy there's no problems. Copying output, just for the records of a working win system, and closing the issue.
```
||| no policy found
||| added context bar
||| finding file [foo.ini]
||| checking [C:\Users\yo\Documents\GitHub\foo.ini] (pwd)
||| checking [C:\Users\yo\AppData\Roaming\yarp\config\robots\default] (robot YARP_CONFIG_HOME)
||| checking [C:\Users\yo\AppData\Roaming\yarp\robots\default] (robot YARP_DATA_HOME)
||| checking [C:\ProgramData\yarp\robots\default] (robot YARP_CONFIG_DIRS)
||| checking [C:\Program Files (x86)\iCub\share\iCub\robots\default] (robot YARP_DATA_DIRS)
||| checking [C:\Program Files (x86)\robotology\YARP\share\yarp\robots\default] (robot YARP_DATA_DIRS)
||| checking [C:\Program Files (x86)\iCub\share\iCub\config\path.d] (robot path.d YARP_DATA_DIRS)
||| checking [C:\Program Files (x86)\robotology\YARP\share\yarp\config\path.d] (robot path.d YARP_DATA_DIRS)
||| checking [C:\Users\yo\AppData\Roaming\yarp\config\contexts\bar] (context YARP_CONFIG_HOME)
||| checking [C:\Users\yo\AppData\Roaming\yarp\contexts\bar] (context YARP_DATA_HOME)
||| checking [C:\ProgramData\yarp\contexts\bar] (context YARP_CONFIG_DIRS)
||| checking [C:\Program Files (x86)\iCub\share\iCub\contexts\bar] (context YARP_DATA_DIRS)
||| checking [C:\Program Files (x86)\robotology\YARP\share\yarp\contexts\bar] (context YARP_DATA_DIRS)
||| checking [C:\Users\yo\AppData\Roaming\yarp\config\foo.ini] (YARP_CONFIG_HOME)
||| checking [C:\Users\yo\AppData\Roaming\yarp\foo.ini] (YARP_DATA_HOME)
||| checking [C:\ProgramData\yarp\foo.ini] (YARP_CONFIG_DIRS)
||| checking [C:\Program Files (x86)\iCub\share\iCub\foo.ini] (YARP_DATA_DIRS)
||| checking [C:\Program Files (x86)\robotology\YARP\share\yarp\foo.ini] (YARP_DATA_DIRS)
||| did not find foo.ini
```

closed 2015-01-15 by jgvictores