robotology/yarp #355: static-noace build failing on Ubuntu 13.10 and 14.04 since 2015-01-09

drdanz, 2015-01-16:
See: http://dashboard.icub.org/index.php?project=YARP&date=2015-01-09

I think it might be related to 08ebc2d4643c9572c8e2d019cfef735c2356665d

drdanz, 2015-01-16:
I wasn't able to reproduce this on a chroot, perhaps it is some issue on the testing machine?
CC @mbrunettini

paulfitz, 2015-01-16:
I've been assuming this is just a stray process on the test machine holding on to a port. When the new integration tests fail, clean-up is a *lot* harder.

drdanz, 2015-01-16:
@paulfitz I'm referring to the linking error, not a test failing...

> /usr/bin/ld: ../lib/libYARP_priv_sqlite.a(sqlite3.c.o): undefined reference to symbol 'dlopen@@GLIBC_2.1'

paulfitz, 2015-01-16:
Oh ok like this one http://dashboard.icub.org/viewBuildError.php?onlydeltap&buildid=195538.  Hmm also can't replicate.  Maybe a link order problem?  That can be a problem on static builds.

drdanz, 2015-01-28:
After some tests I believe that it might be a link order problem in static-noace builds, when sqlite is not available on the system... Unfortunately even after uninstalling ``libsqlite3-dev`` I wasn't able to reproduce the issue. Perhaps the build machines are not updated or not using some repository (i.e. proposed/backports/security)?

Anyway, having a look at sqlite code, I noticed that it uses dlopen (``sqlite3.c:30527``), therefore the library should depend on dl.

We have 2 ways to fix this:
1. Adding ``dl`` as dependency of ``YARP_priv_sqlite``
2. Building ``YARP_priv_sqlite`` with ``-DSQLITE_OMIT_LOAD_EXTENSION``

I vote for the 2nd option, since I believe that we don't need SQLite load extension. @paulfitz what do you think?

paulfitz, 2015-01-28:
Yes absolutely, option 2 is perfect. Thanks for tracking this down @drdanz.

commit 610da4d4 (2015-01-28): [sqlite] Build with -DSQLITE_OMIT_LOAD_EXTENSION

Fixes #355
closed 2015-01-28 by drdanz