robotology/yarp #357: Using YARP from Python without superuser privileges

gsaponaro, 2015-01-16:
Hi, the tutorial http://wiki.icub.org/yarpdoc/yarp_swig.html, more specifically section [Compiling YARP language bindings](http://wiki.icub.org/yarpdoc/yarp_swig.html#yarp_swig_configure), says:

    make
    sudo make install   # Optional, not sane for all languages.  Good for python.

I would like to suggest an alternative method for systems where installing the bindings system-wide is not desired:

    # define this environment variable
    export PYTHONPATH=$PYTHONPATH:$YARP_ROOT/bindings/build
    make

We tested a few YARP Python scripts that we are using for the POETICON++ project, using that PYTHONPATH value, and things are working fine so far. Will report if we observe anything different. @paulfitz : I wonder if the comment "Good for python" in the SWIG tutorial means that there are other Python subtleties which really would require us to sudo make install.

paulfitz, 2015-01-16:
Thanks @gsaponaro for taking the time to suggest this. Definitely worth mentioning the option of using PYTHONPATH.  "Good for python" just meant "works for python", not really intended as a value judgement. Frankly that section needs updating with the equivalent of PYTHONPATH for each language.  Could you do me a favor and edit it to your taste for python? You can edit online here:

 * https://github.com/robotology/yarp/edit/master/doc/yarp_swig.dox

You can also edit in your local yarp repo in `doc/yarp_swig.dox`, then run `make dox` (you'll need `doxygen` installed), then view `dox/html/yarp_swig.html` in your browser, then make a PR.

If you don't have time, don't worry about it, I'll get to it eventually :-)

kt10aan, 2015-01-16:
+1 for  ```export PYTHONPATH=$PYTHONPATH:$PATH_TO_YARP_PYTHON_LIB```.

This is the setup we have been using for all our python/YARP needs and we haven't encounter any problems.

paulfitz, 2015-01-16:
thanks @gsaponaro for PR! Leaving this issue open to remind me to tweak build path in documentation (may not be bindings/build)