robotology/yarp #362: Problem with Qt5: Qt5_DIR-NOTFOUND qhile QT5 is fully installed on the machine

serena-ivaldi, 2015-01-17:
I just git pulled and I cannot compile yarp with QT5, as the flag Qt5_DIR is set to Qt5_DIR-NOTFOUND. However, I checked on my machine and Qt5 is installed. I also did apt-get on all packages of Qt5. Is there a particular package that I should check?
I am on Ubuntu 14.

drdanz, 2015-01-18:
Hello Serena,

Are you sure you installed all the qt5 packages listed here?

ýhttp://wiki.icub.org/yarpdoc/dependencies.html

Also which Ubuntu? 14.04 or 14.10? The required packages are different...

Cheers,
 Daniele

From: Serena Ivaldi
Sent: sabato 17 gennaio 2015 12:55
To: robotology/yarp
Reply To: robotology/yarp
Subject: [yarp] Problem with Qt5: Qt5_DIR-NOTFOUND qhile QT5 is fully installed on the machine (#362)

I just git pulled and I cannot compile yarp with QT5, as the flag Qt5_DIR is set to Qt5_DIR-NOTFOUND. However, I checked on my machine and Qt5 is installed. I also did apt-get on all packages of Qt5. Is there a particular package that I should check?
I am on Ubuntu 14.

kt10aan, 2015-01-18:
Did you cleaned the CMakeCache after apt-getting the rest of the Qt5 packages?

In any case, on my Ubuntu 14.04 installation, I just installed Qt5 from the generic installer (on the Qt5 website) in e.g., /usr/local/Qt5 then pointed the Qt5_DIR to /usr/local/Qt5/5.4/gcc_64/lib/cmake. (Just remember to check the Sources box during theQt5 installation!).

gsaponaro, 2015-01-19:
These are the Ubuntu Qt5-related packages that I have installed with apt-get, please check if you have all of them:

qttools5-dev qtdeclarative5-dev qtdeclarative5-controls-plugin qtmultimedia5-dev qtdeclarative5-qtmultimedia-plugin qtquick1-5-dev

See also #263

drdanz, 2015-01-28:
@serena-ivaldi Is this fixed?

drdanz, 2015-01-28:
Anyway, not a YARP issue since Qt5Config.cmake and the other CMake files for Qt5 are supplied by Qt. Closing this...

closed 2015-01-28 by drdanz