robotology/yarp #363: build fails when CBW2_USE_YARP_THRIFT=FALSE

paulfitz, 2015-01-19:
With `CBW2_USE_YARP_THRIFT=FALSE` in cmake settings, make fails as follows on Linux:
```
In file included from /home/paulfitz/cvs/yarp2/src/libYARP_dev/src/modules/RemoteControlBoard/RemoteControlBoard.cpp:30:0:
/home/paulfitz/cvs/yarp2/src/libYARP_dev/src/modules/RemoteControlBoard/stateExtendedReader.hpp:33:23: fatal error: jointData.h: No such file or directory
 #include "jointData.h"
                       ^
compilation terminated.
src/libYARP_dev/CMakeFiles/YARP_dev.dir/build.make:882: recipe for target 'src/libYARP_dev/CMakeFiles/YARP_dev.dir/src/modules/RemoteControlBoard/RemoteControlBoard.cpp.o' failed
make[2]: *** [src/libYARP_dev/CMakeFiles/YARP_dev.dir/src/modules/RemoteControlBoard/RemoteControlBoard.cpp.o] Error 1
CMakeFiles/Makefile2:560: recipe for target 'src/libYARP_dev/CMakeFiles/YARP_dev.dir/all' failed
make[1]: *** [src/libYARP_dev/CMakeFiles/YARP_dev.dir/all] Error 2
Makefile:147: recipe for target 'all' failed
make: *** [all] Error 2
```
The setting of `CBW2_USE_ROS_MSG` doesn't appear to change this.

lornat75, 2015-01-20:
@barbalberto ?

barbalberto, 2015-01-20:
Sorry, my fault. Providing back compatibility for issue #343 I removed that flaag from the code, but not from the cmake. 
In commit 4eb1629947104df249a0a999d51a0f70948c3e21 I removed the option from the cmake because now the extended port is the default and back compatibility is provided ass a run-time check, so there was no need for that option to exists anymore

traversaro, 2015-01-20:
@barbalberto commit https://github.com/robotology/yarp/commit/4eb1629947104df249a0a999d51a0f70948c3e21 broke compilation of yarp from the master branch, see https://travis-ci.org/robotology/yarp/builds/47624892 .

vtikha, 2015-01-20:
Indeed the compilation of YARP is broken. This is the master branch of YARP and again, why was this changed and not tested (at least compiled) beforehand?? This creates a lot of issues especially being away and having to demonstrate various things to EU partners.

barbalberto, 2015-01-20:
The CMakeCache was hiding the problem, fixed now.
Sorry for the inconvenience

vtikha, 2015-01-20:
I confirm, it now compiles.

closed 2015-01-20 by vtikha

barbalberto, 2015-01-20:
get rid of last YARP_MSG defines in code/cmake.
Hope it does not add other errors  :$
(Tried also with clean cache)