robotology/yarp #367: Cannot make connection to 'xmlrpc://ros'

Farinha91, 2015-01-20:
Hi all,

I recently made a service between ROS and YARP with a custom message and it works well =) but when it closes (say, for example, that my server shuts down after replying to one message) it exits with the following:

yarp: Removing input from /unnamed to /hand_comm-1@/yarp_hand_comm_server
yarp: Removing output from /ISRhand/remote/left_arm/rpc:o to /ISRhand/left_arm/rpc:i
yarp: Removing output from /ISRhand/remote/left_arm/command:o to /ISRhand/left_arm/command:i
yarp: Removing input from /ISRhand/left_arm/state:o to /ISRhand/remote/left_arm/state:i
yarp: Removing input from /ISRhand/left_arm/stateExt:o to /ISRhand/remote/left_arm/stateExt:i
yarp: Removing output from /hand_comm-1@/yarp_hand_comm_server to /unnamed

yarp: Failed to find library support for carrier xmlrpc
yarp: Could not find carrier "xmlrpc"
Cannot make connection to 'xmlrpc://ros'
yarp: Failed to find library support for carrier xmlrpc
yarp: Could not find carrier "xmlrpc"
Cannot make connection to 'xmlrpc://ros'
yarp: Failed to find library support for carrier xmlrpc
yarp: Could not find carrier "xmlrpc"
Cannot make connection to 'xmlrpc://ros'

Now, my implementation of the service and type promise are made by:

    Network yarp;

    //+++++++setting up server++++++;
    RpcServer server;
    ISRHand srvtype;
    server.promiseType(srvtype.getType());

    if (!server.open("/hand_comm@/yarp_hand_comm_server")) {
        fprintf(stderr,"Failed to open port\n");
        return 1;
    }

I'm guessing that the moment the connection is stopped the yarp freaks about the servicetype promise. This is more of a whim as it doesn't really affect the program performance but does anyone knows how can this be solved?

Best regards,
André

paulfitz, 2015-01-20:
Those error messages are all about not being able to load the carrier that supports xmlrpc (used to talk to some bits of ROS). Optional carriers don't get linked to the yarp library these days, but are loaded on need, which means that yarp may need extra help finding the libraries (e.g. by setting `YARP_DATA_DIRS`). But if your service is basically working, then presumably it could find the carrier earlier in the program. Will check if there's a problem in the order things get shut down. Thanks for reporting!

paulfitz, 2015-01-20:
@Farinha91 can you leave this issue open until the problem is fixed? Might be a few days. Thanks!

Farinha91, 2015-01-20:
Oh, sorry

randaz81, 2017-03-30:
Uhm.. I think that this does not happen anymore. What do you think @drdanz?