robotology/yarp #368: Issues in YARP_IDL_TO_DIR Macro

drdanz, 2015-01-20:
The ``YARP_IDL_TO_DIR`` macro has a few issues:

1. The ``ALLOW_IDL_GENERATION`` is set by default to ``TRUE`` or ``FALSE`` depending if the generated files exist. Unfortunately this will cause issues if you have several calls of this macro and if the first call finds the required files (for example because they were committed in the repository), this will set the ``FALSE`` value, and if the files for the second call were not found, they are never generated, unless the user manually sets ``ALLOW_IDL_GENERATION``.

2. In order to be able to use the IDL inside YARP, and to handle all the dependencies properly without requiring to commit the generated files together with yarp, we need a way to foresee at build time which files are generated by the ``yarpidl_xxx`` command.

  At the moment this is handled by ``yarp_idl_to_dir`` in this way:

    * A ``place<name>.cmake is generated from a template.
    * ``yarpidl_xxx`` is executed and generates the files in a temporary folder.
    * The place<name>.cmake is executed with ``cmake -P``. This checks which files are generated by ``yarpidl_xxx`` and copies them in the final destination, then generates a <name>.cmake file that sets the ``headers`` and ``sources`` variable.
    * The <name>.cmake file is included in the same ``yarp_idl_to_dir``
    * The variables that contain the names of the generated files are set in the parent scope.
    * A custom_command and a custom_target are added to regenerate the files at build time.

  Unfortunately this works only if the ``yarpidl_xxx`` executable already exists. Therefore this cannot work inside YARP. This means that it is impossible to add a dependency for a target on a generated files, unless we commit the files in the repository.

  (There is also a minor bug, since ``<name>Reply.h`` files generated from ``.srv`` files are included twice inside the ``<name>.cmake``, but this does not cause any real issue)

  If we had a way to know the names of the generated files, everything would be a lot easier, both in YARP and outside YARP.

  I think that it would be good if we could assume that:

  * yarpidl_thrift generates
    * ``src/<name>_IDL.cpp``
    * ``include/<name>_IDL.h``
  * yarpidl_ros (for ``.msg`` generates)
    * ``include/<name>.h``
  * yarpidl_ros (for ``.srv`` generates)
    * ``include/<name>.h``
    * ``include/<name>Reply.h``

  But unfortunately sometimes ``yarpidl_ros`` generates some extra files (i.e. ``Header.h`` and ``TickTime.h`` and there is no way to foresee if they will be generated without parsing the file)
Perhaps these standard messages could just be installed somewhere with YARP?

3. If there is a dependency between 2 ros message files, these could be executed in the wrong order on a parallel build, and it might fail. I don't know if internally dependencies are actually used, or if it is just necessary to know the header and the class name, but it would be good if they could be generated in any order (the compiler will ensure that you have all the required files).

drdanz, 2015-01-21:
After some more tests I realized that ``yarpidl_ros`` also writes "wrappers" when the file is in a subfolder, therefore

  * yarpidl_thrift generates
    * ``src/<name>_IDL.cpp``
    * ``include/<name>_IDL.h``
  * yarpidl_ros (for ``.msg`` generates)
    * ``include/<path>/<name>.h``
    * (``include/<path>/<clean_path>_<name>.h``)
  * yarpidl_ros (for ``.srv`` generates)
    * ``include/<path>/<name>.h``
    * (``include/<path>/<clean_path>_<name>.h``)
    * ``include/<path>/<name>Reply.h``
    * (``include/<path>/<clean_path>_<name>Reply.h``)

This can be handled in CMake just by checking if there is a path.

Yeshasvitvs, 2017-02-11:
@drdanz any update on this ?