robotology/yarp #374: OS::NodeTest failed on travis

paulfitz, 2015-01-22:
`NodeTest` failed on travis, without code change - it may be revealing a race?

  * https://travis-ci.org/robotology/yarp/jobs/47929078

traversaro, 2016-05-19:
Another instance of this problem occurred in https://travis-ci.org/robotology/yarp/jobs/131403762 : 
~~~
      Start 18: OS::NodeTest
18/65 Test #18: OS::NodeTest .......................***Exception: Other  0.41 sec
yarp: 0 | NodeTest: check support for various experimental node/topic/service schemes
yarp: 0 | NodeTest:   [node name matches for /foo] passed ok
yarp: 0 | NodeTest:   [nested name matches for /foo] passed ok
yarp: 0 | NodeTest:   [category matches for /foo] passed ok
yarp: 0 | NodeTest:   [node name matches for /topic@/foo] passed ok
yarp: 0 | NodeTest:   [nested name matches for /topic@/foo] passed ok
yarp: 0 | NodeTest:   [category matches for /topic@/foo] passed ok
yarp: 0 | NodeTest:   [node name matches for /topic-@/foo] passed ok
yarp: 0 | NodeTest:   [nested name matches for /topic-@/foo] passed ok
yarp: 0 | NodeTest:   [category matches for /topic-@/foo] passed ok
yarp: 0 | NodeTest:   [node name matches for /topic-1@/foo] passed ok
yarp: 0 | NodeTest:   [nested name matches for /topic-1@/foo] passed ok
yarp: 0 | NodeTest:   [category matches for /topic-1@/foo] passed ok
yarp: 0 | NodeTest:   [node name matches for /foo=/topic] passed ok
yarp: 0 | NodeTest:   [nested name matches for /foo=/topic] passed ok
yarp: 0 | NodeTest:   [category matches for /foo=/topic] passed ok
yarp: 0 | NodeTest:   [node name matches for /foo=+/topic] passed ok
yarp: 0 | NodeTest:   [nested name matches for /foo=+/topic] passed ok
yarp: 0 | NodeTest:   [category matches for /foo=+/topic] passed ok
yarp: 0 | NodeTest:   [node name matches for /foo=+1/topic] passed ok
yarp: 0 | NodeTest:   [nested name matches for /foo=+1/topic] passed ok
yarp: 0 | NodeTest:   [category matches for /foo=+1/topic] passed ok
yarp: 0 | NodeTest:   [node name matches for /foo-#/topic] passed ok
yarp: 0 | NodeTest:   [nested name matches for /foo-#/topic] passed ok
yarp: 0 | NodeTest:   [category matches for /foo-#/topic] passed ok
yarp: 0 | NodeTest:   [node name matches for /foo+#/topic] passed ok
yarp: 0 | NodeTest:   [nested name matches for /foo+#/topic] passed ok
yarp: 0 | NodeTest:   [category matches for /foo+#/topic] passed ok
yarp: 0 | NodeTest: most basic node test
yarp(1677fa59): register contact /test=-/p1
yarp: Port /test=-/p1 active at tcp://localhost:10002
yarp(1677fa59): register contact /test=+/p2
yarp: Port /test=+/p2 active at tcp://localhost:10003
yarp(1677fa59): register contact /test
yarp: Port /test active at tcp://localhost:10004
yarp: 0 | NodeTest:   [found /p1] passed ok
yarp: 0 | NodeTest:   [failed to find non-existent /p9] passed ok
yarp: 0 | NodeTest: most basic nodes test
yarp(1677fa59): register contact /test=-/p1
yarp: Port /test=-/p1 active at tcp://localhost:10002
yarp(1677fa59): register contact /test=-/p2
yarp: Port /test=-/p2 active at tcp://localhost:10005
yarp(1677fa59): register contact /test2=+/p3
yarp: Port /test2=+/p3 active at tcp://localhost:10006
yarp(1677fa59): register contact /test
yarp: Port /test active at tcp://localhost:10004
yarp(1677fa59): register contact /test2
yarp: Port /test2 active at tcp://localhost:10007
yarp: 0 | NodeTest:   [found /p1] passed ok
yarp: 0 | NodeTest:   [failed to find non-existent /p9] passed ok
yarp: 0 | NodeTest:   [found /p3] passed ok
yarp(1677fa59): working on connection /test=-/p1 to /test2=+/p3 (connect)
yarp(1677fa59): query name /test=-/p1
yarp(1677fa59): query name /test2=+/p3
yarp(1677fa59): ** asking tcp://test=-/p1: [add] "tcp://test2=+/p3"
yarp(221c5c71): TextCarrier::expectSenderSpecifier
yarp(221c5c71): query name /test2=+/p3
yarp(425e7693): Port /test2=+/p3 starting up, flushing routes /test=-/p1->*->/test2=+/p3
yarp: Receiving input from /test=-/p1 to /test2=+/p3 using tcp
yarp(425e7693): ConnectionState::expectIndex for /test=-/p1->tcp->/test2=+/p3
yarp: Sending output from /test=-/p1 to /test2=+/p3 using tcp
yarp(1677fa59): query name /test=-/p1
yarp(1677fa59): query name /test=-/p1
yarp(221c5c71): TextCarrier::expectSenderSpecifier
yarp(1677fa59): query name /test2=+/p3
yarp(1677fa59): query name /test2=+/p3
[... 73 lines cut ...]
yarp(1677fa59): asking tcp://test1=+/p1: [list] [out] "/test2=-/p1"
yarp(5686222): TextCarrier::expectSenderSpecifier
yarp: 0 | NodeTest:   [auto connect working] passed ok
yarp(425e7693): Total message length: 10
yarp: Removing input from /test1=+/p1 to /test2=-/p1
yarp: Removing output from /test1=+/p1 to /test2=-/p1
yarp: 0 | NodeTest: direction test
yarp(1677fa59): register contact /a/topic+@/node1
yarp(1677fa59): register contact /node1
yarp: Port /node1 active at tcp://localhost:10012
yarp: Port /a/topic+@/node1 active at tcp://localhost:10011
yarp(1677fa59): register contact /a/topic-@/node2
yarp(1677fa59): register contact /node2
yarp: Port /node2 active at tcp://localhost:10014
yarp(fa69144b): working on connection /a/topic+@/node1 to /a/topic-@/node2 (check)
yarp(fa69144b): query name /a/topic+@/node1
yarp: Port /a/topic-@/node2 active at tcp://localhost:10013
yarp(1677fa59): working on connection /a/topic+@/node1 to /a/topic-@/node2 (check)
yarp(1677fa59): query name /a/topic+@/node1
yarp(1677fa59): query name /a/topic-@/node2
yarp(1677fa59): asking tcp://a/topic+@/node1: [list] [out] "/a/topic-@/node2"
yarp(cd6b800e): TextCarrier::expectSenderSpecifier
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_S_create
~~~

traversaro, 2016-05-26:
Again : https://travis-ci.org/robotology/yarp/jobs/133138935 .

Tobias-Fischer, 2016-08-03:
This should be fixed now @traversaro @drdanz

drdanz, 2016-08-03:
Are you sure? Let's wait a few days before closing it...

traversaro, 2016-08-03:
As far as I remember, this problems was reproducible with a:
~~~
ctest --output-on-failure --repeat-until-fail 1000 -R Node
~~~ 
But I had to wait ~400/500 iterations before getting one failure, on my PC with a random load.

Tobias-Fischer, 2016-08-04:
Indeed, this still happens ..

drdanz, 2016-08-04:
It would be really useful if we managed to produce a core dump.

Tobias-Fischer, 2016-08-04:
There you go :)
```
[Current thread is 1 (Thread 0x7f82af7f5700 (LWP 13514))]

Thread 6 (Thread 0x7f82ad7f1700 (LWP 13515)):
#0  0x00007f82b19e77ad in accept () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f82b1d14511 in ACE_SOCK_Acceptor::accept(ACE_SOCK_Stream&, ACE_Addr*, ACE_Time_Value*, bool, bool) const () from /usr/lib/libACE-6.3.3.so
#2  0x00007f82b0e56752 in yarp::os::impl::SocketTwoWayStream::open (this=this@entry=0x7f828c002950, acceptor=...) at /home/icub/robot/yarp/src/libYARP_OS/src/SocketTwoWayStream.cpp:64
#3  0x00007f82b0e5bbe5 in yarp::os::impl::TcpFace::read (this=0x1dbbe70) at /home/icub/robot/yarp/src/libYARP_OS/src/TcpFace.cpp:87
#4  0x00007f82b0de6f5d in yarp::os::impl::PortCore::run (this=0x1beb4e0) at /home/icub/robot/yarp/src/libYARP_OS/src/PortCore.cpp:184
#5  0x00007f82b0e5e7c5 in theExecutiveBranch (args=0x1beb4e0) at /home/icub/robot/yarp/src/libYARP_OS/src/ThreadImpl.cpp:96
#6  0x00007f82b1cf1f56 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.3.3.so
#7  0x00007f82b19de6fa in start_thread (arg=0x7f82ad7f1700) at pthread_create.c:333
#8  0x00007f82b0480b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 5 (Thread 0x7f82aeff4700 (LWP 13512)):
#0  0x00007f82b19e77ad in accept () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f82b1d14511 in ACE_SOCK_Acceptor::accept(ACE_SOCK_Stream&, ACE_Addr*, ACE_Time_Value*, bool, bool) const () from /usr/lib/libACE-6.3.3.so
#2  0x00007f82b0e56752 in yarp::os::impl::SocketTwoWayStream::open (this=this@entry=0x7f82a0002950, acceptor=...) at /home/icub/robot/yarp/src/libYARP_OS/src/SocketTwoWayStream.cpp:64
#3  0x00007f82b0e5bbe5 in yarp::os::impl::TcpFace::read (this=0x1d21010) at /home/icub/robot/yarp/src/libYARP_OS/src/TcpFace.cpp:87
#4  0x00007f82b0de6f5d in yarp::os::impl::PortCore::run (this=0x1becf20) at /home/icub/robot/yarp/src/libYARP_OS/src/PortCore.cpp:184
#5  0x00007f82b0e5e7c5 in theExecutiveBranch (args=0x1becf20) at /home/icub/robot/yarp/src/libYARP_OS/src/ThreadImpl.cpp:96
#6  0x00007f82b1cf1f56 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.3.3.so
#7  0x00007f82b19de6fa in start_thread (arg=0x7f82aeff4700) at pthread_create.c:333
#8  0x00007f82b0480b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 4 (Thread 0x7f82977fe700 (LWP 13513)):
#0  0x00007f82b19e77ad in accept () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f82b1d14511 in ACE_SOCK_Acceptor::accept(ACE_SOCK_Stream&, ACE_Addr*, ACE_Time_Value*, bool, bool) const () from /usr/lib/libACE-6.3.3.so
#2  0x00007f82b0e56752 in yarp::os::impl::SocketTwoWayStream::open (this=this@entry=0x7f829c002950, acceptor=...) at /home/icub/robot/yarp/src/libYARP_OS/src/SocketTwoWayStream.cpp:64
#3  0x00007f82b0e5bbe5 in yarp::os::impl::TcpFace::read (this=0x1dbc950) at /home/icub/robot/yarp/src/libYARP_OS/src/TcpFace.cpp:87
#4  0x00007f82b0de6f5d in yarp::os::impl::PortCore::run (this=0x1d1f0d0) at /home/icub/robot/yarp/src/libYARP_OS/src/PortCore.cpp:184
#5  0x00007f82b0e5e7c5 in theExecutiveBranch (args=0x1d1f0d0) at /home/icub/robot/yarp/src/libYARP_OS/src/ThreadImpl.cpp:96
#6  0x00007f82b1cf1f56 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.3.3.so
#7  0x00007f82b19de6fa in start_thread (arg=0x7f82977fe700) at pthread_create.c:333
#8  0x00007f82b0480b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 3 (Thread 0x7f82adff2700 (LWP 13511)):
#0  0x00007f82b19e77ad in accept () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f82b1d14511 in ACE_SOCK_Acceptor::accept(ACE_SOCK_Stream&, ACE_Addr*, ACE_Time_Value*, bool, bool) const () from /usr/lib/libACE-6.3.3.so
#2  0x00007f82b0e56752 in yarp::os::impl::SocketTwoWayStream::open (this=this@entry=0x7f8298002950, acceptor=...) at /home/icub/robot/yarp/src/libYARP_OS/src/SocketTwoWayStream.cpp:64
#3  0x00007f82b0e5bbe5 in yarp::os::impl::TcpFace::read (this=0x1d21530) at /home/icub/robot/yarp/src/libYARP_OS/src/TcpFace.cpp:87
---Type <return> to continue, or q <return> to quit---
#4  0x00007f82b0de6f5d in yarp::os::impl::PortCore::run (this=0x1d23be0) at /home/icub/robot/yarp/src/libYARP_OS/src/PortCore.cpp:184
#5  0x00007f82b0e5e7c5 in theExecutiveBranch (args=0x1d23be0) at /home/icub/robot/yarp/src/libYARP_OS/src/ThreadImpl.cpp:96
#6  0x00007f82b1cf1f56 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.3.3.so
#7  0x00007f82b19de6fa in start_thread (arg=0x7f82adff2700) at pthread_create.c:333
#8  0x00007f82b0480b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 2 (Thread 0x7f82b215d740 (LWP 13468)):
#0  0x00007f82b09e4f10 in operator delete(void*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007f82b0daf9fd in __gnu_cxx::new_allocator<char>::deallocate (this=0x7fff3bc6ce60, __p=<optimised out>) at /usr/include/c++/5/ext/new_allocator.h:110
#2  __gnu_cxx::__alloc_traits<std::allocator<char> >::deallocate (__a=..., __n=<optimised out>, __p=<optimised out>) at /usr/include/c++/5/ext/alloc_traits.h:185
#3  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_destroy (__size=<optimised out>, this=0x7fff3bc6ce60) at /usr/include/c++/5/bits/basic_string.h:185
#4  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose (this=0x7fff3bc6ce60) at /usr/include/c++/5/bits/basic_string.h:180
[... cut ...]
#26 0x00000000004e73af in ConnectThread::run (this=0x1f0a7b0) at /home/icub/robot/yarp/src/libYARP_serversql/src/ConnectThread.h:53
#27 0x00007f82b0e5e7c5 in theExecutiveBranch (args=0x1f0a820) at /home/icub/robot/yarp/src/libYARP_OS/src/ThreadImpl.cpp:96
#28 0x00007f82b1cf1f56 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.3.3.so
#29 0x00007f82b19de6fa in start_thread (arg=0x7f82af7f5700) at pthread_create.c:333
#30 0x00007f82b0480b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

Tobias-Fischer, 2016-08-04:
Guess there's a `lock()` / `unlock()` missing in `NameServiceOnTriples::apply()` (https://github.com/robotology/yarp/blob/devel/src/libYARP_serversql/src/NameServiceOnTriples.cpp). Should be relatively easy to fix for someone who knows the code (which seems to be @drdanz from the history)

drdanz, 2016-08-05:
@Tobias-Fischer Thanks a lot! I've been running the test for hours without being able to reproduce it!
I don't know the code well, but I have a branch where I was trying to fix errors reported by valgrind using the helgrind and drd tools. I'm pretty sure there were at least a couple of fixes in NameServicesOnTriples, so perhaps I just need to cherry-pick the relative commits.

drdanz, 2016-08-08:
Should be fixed in 80e9f26d66241496b7f013ba90749d3f64eaea44

PR #860 (2016-08-08): Try to fix NodeTest race condition
`YARP_serversql/NameServiceOnTriples`: Protect concurrent access to internal variables using a mutex

Fixes several race conditions reported by valgrind

Might fix #374 

CC @Tobias-Fischer
<a href='#crh-start'></a><a href='#crh-data-%7B%22processed%22%3A%20%5B%22https%3A//github.com/robotology/yarp/pull/860%23discussion_r73843846%22%2C%20%22https%3A//github.com/robotology/yarp/pull/860%23issuecomment-238182507%22%2C%20%22https%3A//github.com/robotology/yarp/pull/860%23discussion_r73850323%22%2C%20%22https%3A//github.com/robotology/yarp/pull/860%23issuecommen [... cut]
commit 80e9f26d (2016-08-08): libYARP_serversql/NameServiceOnTriples: Protect concurrent access to internal variables using a mutex

Fixes several race conditions reported by valgrind

Might fix #374
closed 2016-08-08 by drdanz