robotology/yarp #375: Warnings while compiling yarp with SWIG and Python support

gsaponaro, 2015-01-23:
With Ubuntu 14.04, yarp 47d61279ae294ca5ad7780a947b707e4181786e0, SWIG 2.0.11 and CREATE_PYTHON=ON I noticed the following warnings:

```
[ 88%] Swig source
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:98: Warning 389: operator[] ignored (consider using %extend)
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:99: Warning 389: operator[] ignored (consider using %extend)
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:141: Warning 321: 'hash' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Contactable.h:52: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Contactable.h:62: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Contactable.h:74: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Contactable.h:74: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:53: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:57: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:57: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/RateThread.h:230: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/RateThread.h:230: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/RateThread.h:230: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_sig/include/yarp/sig/Sound.h:65: Warning 321: 'set' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_sig/include/yarp/sig/Sound.h:65: Warning 321: 'set' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_sig/include/yarp/sig/Vector.h:192: Warning 389: operator[] ignored (consider using %extend)
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_sig/include/yarp/sig/Vector.h:202: Warning 389: operator[] ignored (consider using %extend)
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_sig/include/yarp/sig/Vector.h:447: Warning 389: operator[] ignored (consider using %extend)
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_sig/include/yarp/sig/Vector.h:455: Warning 389: operator[] ignored (consider using %extend)
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/IConfig.h:44: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:57: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_dev/include/yarp/dev/PolyDriver.h:79: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h:203: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h:218: Warning 321: 'open' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/bindings/yarp.i:537: Warning 302: Identifier 'TypedReaderCallbackImageMono' redefined (ignored) (Renamed from 'TypedReaderCallback< yarp::sig::Sound >'),
/home/gsaponaro/NOBACKUP/yarp/bindings/yarp.i:528: Warning 302: previous definition of 'TypedReaderCallbackImageMono' (Renamed from 'TypedReaderCallback< yarp::sig::ImageOf< yarp::sig::PixelMono > >').
/home/gsaponaro/NOBACKUP/yarp/bindings/yarp.i:978: Warning 321: 'set' conflicts with a built-in name in python
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/BufferedPort.h:1: Warning 401: Base class 'TypedReaderCallback< yarp::sig::Sound >' undefined.
/home/gsaponaro/NOBACKUP/yarp/bindings/yarp.i:537: Warning 401: 'TypedReaderCallback< yarp::sig::Sound >' must be defined before it is used as a base class.
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/PortablePair.h:23: Warning 403: Class 'yarp::os::PortablePairBase' might be abstract, no constructors generated,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34: Warning 403: Method yarp::os::Portable::read(yarp::os::ConnectionReader &) might not be implemented.
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:200: Warning 314: 'is' is a python keyword, renaming to '_is'
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:200: Warning 314: 'is' is a python keyword, renaming to '_is'
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:200: Warning 314: 'is' is a python keyword, renaming to '_is'
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:202: Warning 314: 'is' is a python keyword, renaming to '_is'
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:202: Warning 314: 'is' is a python keyword, renaming to '_is'
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:202: Warning 314: 'is' is a python keyword, renaming to '_is'
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Value.h:247: Warning 509: Overloaded method yarp::os::Value::check(yarp::os::ConstString const &) const effectively ignored,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:80: Warning 509: as it is shadowed by check(yarp::os::ConstString const &) const.
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Value.h:253: Warning 509: Overloaded method yarp::os::Value::findGroup(yarp::os::ConstString const &) const effectively ignored,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:113: Warning 509: as it is shadowed by findGroup(yarp::os::ConstString const &) const.
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:49: Warning 512: Overloaded method yarp::os::AbstractContactable::asPort() const ignored,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:42: Warning 512: using non-const method yarp::os::AbstractContactable::asPort() instead.
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:57: Warning 509: Overloaded method yarp::os::AbstractContactable::open(yarp::os::Contact const &) effectively ignored,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Contactable.h:74: Warning 509: as it is shadowed by open(yarp::os::Contact const &).
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:53: Warning 509: Overloaded method yarp::os::AbstractContactable::open(yarp::os::ConstString const &) effectively ignored,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Contactable.h:62: Warning 509: as it is shadowed by open(yarp::os::ConstString const &).
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Contactable.h:62: Warning 509: Overloaded method open(yarp::os::ConstString const &) effectively ignored,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:53: Warning 509: as it is shadowed by yarp::os::AbstractContactable::open(yarp::os::ConstString const &).
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Contactable.h:74: Warning 509: Overloaded method open(yarp::os::Contact const &,bool) effectively ignored,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/AbstractContactable.h:57: Warning 509: as it is shadowed by yarp::os::AbstractContactable::open(yarp::os::Contact const &,bool).
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Property.h:77: Warning 509: Overloaded method yarp::os::Property::check(yarp::os::ConstString const &) const effectively ignored,
/home/gsaponaro/NOBACKUP/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:80: Warning 509: as it is shadowed by check(yarp::os::ConstString const &) const.
[... cut ...]
/home/gsaponaro/NOBACKUP/yarp/build/bindings/yarpPYTHON_wrap.cxx:67567:12: warning: deleting object of abstract class type ‘yarp::dev::CartesianEvent’ which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
     delete arg1;
            ^
/home/gsaponaro/NOBACKUP/yarp/build/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_delete_GazeEvent(PyObject*, PyObject*)’:
/home/gsaponaro/NOBACKUP/yarp/build/bindings/yarpPYTHON_wrap.cxx:71385:12: warning: deleting object of abstract class type ‘yarp::dev::GazeEvent’ which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
     delete arg1;
            ^
```

paulfitz, 2015-01-23:
@gsaponaro is there a particular warning that troubles you? The less "swiggy" ones like those about non-virtual destructors look important for example. The "swiggy" ones are annoying but haven't been a priority so far that I've heard?

gsaponaro, 2015-01-23:
None of those trouble me or my work. Sorry for not being clear initially. Just wanted to report these, feel free to postpone or close.

paulfitz, 2015-01-23:
Ok. Thanks for reporting. Swig generates a lot of warnings for each language that we've just been living with up to now. I'll rename this issue slightly and leave it around for anyone who wants to +1 dealing with them.

pattacini, 2015-09-02:
It's very unlikely all these warnings will be fixed.
Closing.

closed 2015-09-02 by pattacini

traversaro, 2015-09-02:
If anyone wants to fix them, they can be easily fixed using the `%ignore` swig directive (not volunteering for that anyway! :D ).