robotology/yarp #378: coveralls

lornat75, 2015-01-26:
Cool feature. It causes some noise though. Any way to improve how the way it logs messages?

drdanz, 2015-01-26:
Which messages do you mean? The comments added to the pull requests?

barbalberto, 2015-01-26:
BTW, what does it means "coverall decreased/increased"?

lornat75, 2015-01-26:
@drdanz yes

drdanz, 2015-01-26:
@barbalberto when the tests are run on travis, the lines of code executed by the tests are saved and then reported to the coveralls.io website. From there you can browse the source and see which parts of the files are executed and how many times, for example see https://coveralls.io/files/439326758
Green lines are executed at least once (you can see how many times in the number on the right) and red lines are never executed (that means your tests do not cover a case that ends up executing that line.
For example in that file you can see that the setTypeOfServices method is not executed.

The total number of lines executed is calculated and a percentage in comparison of the not executed ones is generated. This percentage increases/decreases depending on the changes you commit. For example if you add a method and do not add a unit test, the percentage should decrease, if you add a unit test to cover one case not yet tested it should increase.

drdanz, 2015-01-26:
@lornat75 Unfortunately the only setting on coveralls.io is "Comment on pull requests?".
I think that the comment is confusing because coveralls does not wait for all the build jobs to end before posting... 
We have 3 different builds that submit to coveralls.io, and they execute different parts of code depending on which switch are turned on, therefore I believe this confuses a coveralls.
I think this confuses the website as well because the coverage seems to change for a lot of files on every commit.
 
Perhaps we should keep it only for one build? @paulfitz what do you think?

drdanz, 2015-01-26:
Also I think they have some issue with pull requests from branches from other repositories, they seem to appear as if they were commits on master, so I don't know if the increasing/decreasing percentage is actually compared to the last commit on master or to the last coverage check that appears in master.

barbalberto, 2015-01-26:
Thanks @drdanz, it make more sense now. Reaaly good feature to have.

paulfitz, 2015-01-26:
@drdanz I like it, current settings ok for me, doesn't seem like noise. The numbers are kind of strange I guess for the reasons you mention. I think we'd better restrict it to exactly one of the travis jobs to get consistency.

jgvictores, 2015-01-28:
I like it as an additional feature/tool.
@drdanz Is the .coveralls.yml file required or used for anything?

drdanz, 2015-01-28:
@jgvictores I'm not 100% sure that it is actually required, but it contains the token used by coveralls to associate a submission to the corresponding repository...

jgvictores, 2015-01-28:
@drdanz From what I've tested, GitHub pushes the branch and you do not need to cite the token explicitly. For local pushes, you can 'coveralls-lcov --repo-token <your token here> coverage.info' once your coverage.info is prepared (can be done in the --source-encoding line too). Not suggesting removing/maintaining the file itself, just trying to nail down what is actually necessary for the setup.

closed 2015-06-29 by drdanz