robotology/yarp #381: Yarpmanager crash report

Ommac, 2015-01-27:
We (me and @pattacini) have experienced a yarpmanager double crash with the following yarp installation during a demo rehearsal. A system restart has been necessary to restore system operability.

```
icub@icub16:/usr/local/src/robot/yarp (master)$ git log -1
commit 47d61279ae294ca5ad7780a947b707e4181786e0
Merge: a077aab 3118211
Author: Daniele E. Domenichelli <daniele.domenichelli@gmail.com>
Date:   Thu Jan 22 19:00:59 2015 +0100

    Merge pull request #373 from robotology/set_envelope
    
    [swig] fix translation of Contactable::setEnvelope, fixes #372
icub@icub16:/usr/local/src/robot/yarp (master)$ git status
# On branch master
# Untracked files:
#   (use "git add <file>..." to include in what will be committed)
#
#	CMakeLists.txt.user
#	conf/_icub04.conf
#	conf/yarp.conf
#	conf/yarp_namespace.conf
#	src/yarpscope-gtk/examples/test_cart.xml
#	src/yarpscope-gtk/examples/test_cart1.xml
#	src/yarpscope-gtk/examples/test_cart2.xml
#	src/yarpscope-gtk/examples/test_cart_torso.xml
nothing added to commit but untracked files present (use "git add" to track)

```

Scenario details:

- Opened gyarpmanager
- Opened, run and connected icubstartup and face_expressions. This worked fine 
- Opened an application (Interactive_Objects_Learning_with_Skin_GUIs)
- Opened application's xml in gedit from yarpmanager
- Modified and saved the xml
- Closed gedit
- Issued a right click + reopen on the Interactive_Objects_Learning_with_Skin_GUIs app
- Yarpmanager freezed, we waited few minutes, no recovery occurred
- Issued ctrl-C to close yarpmanager
- Re-launched yarpmanager from terminal
- Opened and refreshed icubstartup and face_expressions
- Opened, run and connected skin_gui_all
- skin_gui_all worked fine, but yarpmanager crashed again and its window closed

Yarpmanager's output for the 2nd crash is the following:

```
icub@icub16:~$ yarpmanager
||| did not find ymanager.ini
yarp: cannot read from ymanager.ini
||| did not find ymanager.ini
/home/icub/.local/share/yarp/applications/iol_skin.xml
*** glibc detected *** yarpmanager: double free or corruption (fasttop): 0x00007ffcbc109f40 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x76aa6)[0x7ffced459aa6]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6c)[0x7ffced45e84c]
/usr/lib/x86_64-linux-gnu/tls/libnvidia-tls.so.304.117(+0x1cd1)[0x7ffcea174cd1]
======= Memory map: ========
00400000-0053d000 r-xp 00000000 00:14 6302787                            /usr/local/src/robot/yarp/build-x86_64/bin/yarpmanager
0073c000-00744000 rw-p 0013c000 00:14 6302787                            /usr/local/src/robot/yarp/build-x86_64/bin/yarpmanager
009a5000-0208e000 rw-p 00000000 00:00 0                                  [heap]
7ffcbb7ff000-7ffcbb800000 ---p 00000000 00:00 0 
7ffcbb800000-7ffcbc000000 rw-p 00000000 00:00 0 
7ffcbc000000-7ffcbc114000 rw-p 00000000 00:00 0 
7ffcbc114000-7ffcc0000000 ---p 00000000 00:00 0 
7ffcc0a65000-7ffcc0a66000 ---p 00000000 00:00 0 
7ffcc0a66000-7ffcc1266000 rw-p 00000000 00:00 0 
7ffcc1266000-7ffcc1267000 ---p 00000000 00:00 0 
7ffcc1267000-7ffcc1a67000 rw-p 00000000 00:00 0 
[... 100 lines cut ...]
7ffcd65d2000-7ffcd65d3000 rw-p 00006000 00:14 1443343                    /usr/local/src/robot/Qt5.2.1/5.2.1/gcc_64/plugins/imageformats/libqgif.so
7ffcd65d3000-7ffcd65d5000 r-xp 00000000 08:03 2228395                    /usr/lib/x86_64-linux-gnu/pango/1.6.0/modules/pango-basic-fc.so
7ffcd65d5000-7ffcd67d4000 ---p 00002000 08:03 2228395                    /usr/lib/x86_64-linux-gnu/pango/1.6.0/modules/pango-basic-fc.so
7ffcd67d4000-7ffcd67d5000 r--p 00001000 08:03 2228395                    /usr/lib/x86_64-linux-gnu/pango/1.6.0/modules/pango-basic-fc.so
7ffcd67d5000-7ffcd67d6000 rw-p 00002000 08:03 2228395                    /usr/lib/x86_64-linux-gnu/pango/1.6.0/modules/pango-basic-fc.so
7ffcd67d6000-7ffcd67d9000 r-xp 00000000 08:03 131222                     /lib/x86_64-linux-gnu/libgpg-error.so.0.8.0
7ffcd67d9000-7ffcd69d8000 ---p 00003000 08:03 131222                     /lib/x86_64-linux-gnu/libgpg-error.so.0.8.0
7ffcd69d8000-7ffcd69d9000 rw-p 00002000 08:03 131222                     /lib/x86_64-linux-gnu/libgpg-error.so.0.8.0
7ffcd69d9000-7ffcd69ea000 r-xp 00000000 08:03 1970392                    /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.0.0
7ffcd69ea000-7ffcd6be9000 ---p 00011000 08:03 1970392                    /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.0.0
7ffcd6be9000-7ffcd6bea000 r--p 00010000 08:03 1970392                    /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.0.0
7ffcd6bea000-7ffcd6beb000 rw-p 00011000 08:03 1970392                    /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.0.0
7ffcd6beb000-7ffcd6bfb000 r-xp 00000000 08:03 1970394                    /usr/lib/x86_64-linux-gnu/libtasn1.so.3.1.16
7ffcd6bfb000-7ffcd6dfa000 ---p 00010000 08:03 1970394                    /usr/lib/x86_64-linux-gnu/libtasn1.so.3.1.16
7ffcd6dfa000-7ffcd6dfb000 r--p 0000f000 08:03 1970394                    /usr/lib/x86_64-linux-gnu/libtasn1.so.3.1.16
7ffcd6dfb000-7ffcd6dfc000 rw-p 00010000 08:03 1970394                    /usr/lib/x86_64-linux-gnu/libtasn1.so.3.1.16
7ffcd6dfc000-7ffcd6dfe000 r-xp 00000000 08:03 131157                     /lib/x86_64-linux-gnu/libutil-2.13.so
7ffcd6dfe000-7ffcd6ffd000 ---p 00002000 08:03 131157                     /lib/x86_64-linux-gnu/libutil-2.13.so
7ffcd6ffd000-7ffcd6ffe000 r--p 00001000 08:03 131157                     /lib/x86_64-linux-gnu/libutil-2.13.so
7ffcd6ffe000-7ffcd6fff000 rw-p 00002000 08:03 131157                     /lib/x86_64-linux-gnu/libutil-2.13.so
7ffcd6fff000-7ffcd700f000 r-xp 00000000 08:03 1969571                    /usr/lib/x86_64-linux-gnu/libavahi-client.so.3.2.9Aborted
icub@icub16:~$ 

```
Unfortunately, we didn't save the output for the first crash

drdanz, 2015-01-27:
That's hardly related to the commit you pointed (that is just adding some bindings for some functions), and looks like an issue with the nvidia driver (confirmed by the fact that you had to reboot to fix it)
Are you sure you didn't change something in your setup or updated/upgraded some nvidia or kernel package?

pattacini, 2015-01-27:
The setup is the one of the Bluey; we have no info about the current status of such a Nvidia installation. Once back to IIT that needs to be checked.
The problem is not fixed at all; the crash keeps happening randomly.

alecive, 2015-01-27:
What about the gtk version? Is that ok?

pattacini, 2015-01-27:
Not used.

drdanz, 2015-01-28:
Looks like a duplicate of #222 (reported on 5 Sep 2014), am I wrong? So maybe it is not a nvidia driver issue, but some bug in ``yarpmanager-qt``

alecive, 2015-01-28:
The bugs in `yarpmanager` are very difficult to track down. I'd say that they're coming from different parts of the sw, but who knows?

apaikan, 2015-01-28:
@Ommac  and @pattacini 
If you have time, can you please post the result of `ldd yarpmanager`?  
A quick search shows that it might be due to the mix qt4 and qt5 linking.

lornat75, 2015-01-29:
Is is repeatable?

randaz81, 2015-01-29:
We need to allocate a person for a deep debug of the qt-yarpmanager. Similar crashes have been observed by many persons (including myself) since August 2014. Since it happens on different clusters  (e.g. black, blue icub) I'm personally more convinced of a memory bug than a driver/linking issue (but maybe there exists more than a single faulty condition). I remember a demo in which the manager was systematically crashing on the closure event of a module tab. In this case gdb could help to quickly identify the issue.

apaikan, 2015-01-29:
@randaz81 Do you remember which demo?

traversaro, 2015-01-29:
If we suspect a memory issue it may be worth to do a quick check using
valgrind , even in normal conditions.

randaz81, 2015-01-29:
try some iterations of following loop:
1) robotInterface + camera + icubstartup (gazecontroller, WBD, cartesian solvers) + face expressions
2) start red ball demo
3) start skin gui + icub gui
4) close 2 and 3
5) start IOL+all viewers (I remember that I was trying the IOL version with the "funny postures", but I think that also the standard version is ok for testing)
6) close 5
7) restart from 2

lornat75, 2015-01-29:
@apaikan: can you take care of this?

apaikan, 2015-01-29:
ok.

Ommac, 2015-01-30:
@drdanz : You are right, the commit I pointed at in the beginning is not likely to be related to the crashes. I cited it just to specify the current version of the yarp installation on Bluey's setup.

I have checked again issue #222. It seems that crash number 1 reported there is similar to this one. In fact, in both cases we obtain the following:
```
*** glibc detected *** gyarpmanager: double free or corruption (out): 0x00007fd13007b4b0 ***
```
However, in #222 the crash happened as I was trying to use the "attach to stdout" functionality, which has not been used here.

@apaikan : I'm not at IIT today. If the Bluey setup is up and running I can do that tomorrow. Is it so? If it is, I could also try the scenario suggested by @randaz81. Please let me know, thank you.

alecive, 2015-06-12:
Shall we close this? It does not seem to be present any more, at least.