robotology/yarp #384: Standard headers for MSVC

drdanz, 2015-01-29:
We have these files for MSVC in YARP:

* ``src/idls/thrift/msvc/inttypes.h``
* ``src/idls/thrift/msvc/stdint.h``
* ``src/libYARP_manager/include/yarp/manager/ymm-dir.h`` (slightly modified version of ``dirent.h``)
* ``src/yarpdataplayer-gtk/include/msvc/dirent.h``
* ``src/yarpdataplayer-qt/msvc/dirent.h``

``inttypes.h`` and ``stdint.h`` are standard C99 headers, ``dirent.h`` is standard posix.

I think we should:

* [ ] Check if these are required for all MSVC version (including newer) and ensure that the compiler version is used when available.
* [ ] Move these files outside from the ``src`` directory (perhaps in ``extern``), so that we can avoid duplicating these files.

paulfitz, 2015-01-29:
Sounds good to me!

drdanz, 2015-11-27:
`cinttypes` and `cstdint` are standard c++11 headers, so we can rely on them as soon as we require c++11

traversaro, 2017-06-29:
FYI: A MIT-licensed dirent compatibility-layer is available at https://github.com/tronkko/dirent , and it has been recently packaged in vcpkg : https://github.com/Microsoft/vcpkg/pull/1345#event-1143223257 .

drdanz, 2017-06-30:
in c++17 dirent should no longer be necessary 😆

traversaro, 2017-07-01:
I guess YARP mantainers in 2024 will be very happy!

drdanz, 2018-02-16:
`cstdint` and `cinttypes` are standard c++11, we can use them and remove these files, since all the supported MSVC versions have them.
Apparently, the only remaining issue is `dirent.h`, we should move it in extern. I think the origin of the file is this: https://github.com/tronkko/dirent

drdanz, 2018-02-16:
Stupid github (or maybe it is the zenhub addon) hiding previous comment, I was pretty sure I wrote this in the past! 😆

drdanz, 2018-02-22:
This is fixed in latest devel.

drdanz, 2018-02-22:
Fixed by 5bf69878c1a6526b182d9838a74636db65719fc6

PR #1551 (2018-02-18): Fixes pre re-license
* Import and use hmac from https://github.com/ogay/hmac (Fixes #1540)
* Import and use dirent from https://github.com/tronkko/dirent (Fixes #384)
* Move devices in the src/devices subfolder or in the plugins folder (Fixes #1543, Fixes #1544, Fixes #1545, Fixes #1546, Fixes #1547, Fixes #1548, Fixes #1549)
* Remove `yarp/dev/IKinectDeviceDriver.h` and `yarp/dev/IOpenNI2DeviceDriver.h` includes
* Several plugins are currently disabled.
* The "serial" device is no longer a plugin.

Changelog will follow since a few changes might be reverted in the next few days.
PR #1555 (2018-02-22): Move thrift in extern and import latest version (0.11.0)
* Import thrift 0.11.0 in extern
* Use thrift from extern
* Port to thrift 0.11
* Improve build system
* Remove embedded thrift version
* Remove msvc version of stdint.h and inttypes.h (Fixes #384)
commit 5bf69878 (2018-02-22): yarpidl_thrift: Use thrift from extern

* Port to thrift 0.11
* Improve build system
* Remove embedded thrift version
* Remove msvc version of stdint.h and inttypes.h (Fixes #384)
* Use bison (>=2.5) and flex to regenerate files if available

squash! yarpidl_thrift: Use thrift from extern
closed 2018-02-22 by drdanz