robotology/yarp #385: qtYarpmanager crashed

alecive, 2015-01-29:
platform: `blackie`
version: `qt`

terminal output:
```
icub@icub15:~$ yarpmanager
||| did not find ymanager.ini
yarp: cannot read from ymanager.ini
||| did not find ymanager.ini
QString::arg: Argument missing: /inspect, /icub/inertial
QString::arg: Argument missing: /inspect, /icub/inertial
Segmentation fault
```

Shall we start adding more verbosity to `qtyarpmanager` in order to better track its behavior down?

francesco-romano, 2015-01-29:
Maybe also compiling it in Debug mode?

alecive, 2015-07-20:
Closing because difficult to reproduce without more information. I'd still add some verbosity to the manager in order to track its behavior down, though. @apaikan

closed 2015-07-20 by alecive