robotology/yarp #387: IPositionControl2::checkMotionDone() over joints sub-set does not work

pattacini, 2015-01-30:
I've tested the malfunction on the gaze controller where the `checkMotionDone()` is called over just the first 3 joints (out of 6).

lornat75, 2015-01-30:
We should reproduce this problem in https://github.com/robotology/robot-testing before fixing it.

barbalberto, 2015-01-30:
Shall we use the iCub_SIM, gazebo or fakebot or similar to reproduce it? The bug could be in the communication or in the driver implementation.
@ugo I guess you found it on the iCub, is it the same also on simulators?

pattacini, 2015-01-30:
I didn't test it on the simulator because the blending of position and velocity commands was not available on iCub_SIM in the past and, however, I presume it is still to be validated.

barbalberto, 2015-02-02:
The description in the header file was wrong so lead to a wrong usage, the function always return a single bool value also when called for a group of joints or all joints.
This is the only function which is asymmetric in the input/output: you ask for 3 joints and get 1 answer only and therefore is easy to forget this behaviour.
As @pattacini was pointing out, we also lose some information in the process, which joint has not complete the movement yet.

pattacini, 2015-02-02:
thanks @barbalberto 
@lornat75, what do you think about asymmetry and loosing of details of the single joints?

pattacini, 2015-02-02:
cc @AnneLaureM @gregoire-pointeau

lornat75, 2015-02-02:
In the meanwhile I have added check for IPositionControl2 in robotology/robot-testing. 

Can we close this specific issue and open another one specific for the design of checkmotiondone?

closed 2015-02-02 by pattacini