robotology/yarp #388: yarpdataplayer-gtk compile failure without opencv

paulfitz, 2015-01-30:
yarpdataplayer-gtk tries and fails to compile if opencv is not present

```
[ 86%] Building CXX object src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/src/main.cpp.o
In file included from /root/yarp2/src/yarpdataplayer-gtk/include/utils.h:32:0,
                 from /root/yarp2/src/yarpdataplayer-gtk/include/main_window.h:26,
                 from /root/yarp2/src/yarpdataplayer-gtk/src/main.cpp:27:
/root/yarp2/src/yarpdataplayer-gtk/include/worker.h:22:16: fatal error: cv.h: No such file or directory
compilation terminated.
```

vtikha, 2015-02-02:
The latest changes seem to have broken the compilation switch (cmake) for the yarpdataplayer. I have both opencv and QT5 installed and yarpdataplayer is not being included in the yarp project. This has been noted on OS X and on Windows.

vtikha, 2015-02-02:
Shouldn't OPENCV be written as: OpenCV?

drdanz, 2015-02-05:
@vtikha Can you please confirm that this is fixed now?

vtikha, 2015-02-05:
No it is not fixed! again...It seems that YARP_USE_OPENCV and subsequently YARP_HAS_OPENCV are not being set properly.

vtikha, 2015-02-05:
e.g: if I put OpenCV_FOUND instead of YARP_HAS_OPENCV it works, but obviously this is not a proper way, it was just to try it out and confirm the above issue.

drdanz, 2015-02-05:
Can you check if YARP_USE_OPENCV is enabled in your CMake cache?

drdanz, 2015-02-06:
@vtikha Sorry, I looked again and everything seems good to me now, let me know when you have some time and we can investigate it on your computer...
Anyway, please ensure that ``YARP_USE_OPENCV`` is enabled, this is a flag that disables OpenCV (and therefore all packages that depend on it), even if it was found by CMake.

pattacini, 2015-02-06:
Anyway, I've got the same problem on Windows.
Cannot find `YARP_USE_OPENCV` within cmake flags though, neither under advanced ones...

drdanz, 2015-02-06:
@vtikha This time should be fixed for real :smile: 
Can you please confirm?

closed 2015-02-06 by drdanz

pattacini, 2015-02-06:
it works on Windows.

vtikha, 2015-02-08:
I confirm it now works for real!

PR #389 (2015-01-30): [yarpdataplayer-gtk] Do not build without opencv
Fixes #388

@paulfitz I cannot test the fix right now (working with the smartphone) Can you please test it and eventually merge it?
If it works, I think that the same needs to be done for the qt version...
commit 3d2473cb (2015-01-30): [yarpdataplayer-gtk] Do not build without opencv

Fixes #388
commit 53b29384 (2015-02-05): Fix copy/paste error

Fixes #388
commit e1576589 (2015-02-06): Ensure that checkandset_dependency() is called for OpenCV

Fixes #388