robotology/yarp #390: checkMotionDone() over joints sub-sets

pattacini, 2015-02-02:
Presently, the `checkMotionDone(const int n_joint, const int *joints, bool *flag)` method returns in `flag` only one bool performing the logic _and_ over the inquired sub-set of joints.

This creates a sort of asymmetry with respect to all other methods that by contrast provide back arrays of returned variables. Moreover, the user looses somehow the detail of which joint is still moving and which has come to a complete stop.

Would it be worth making the design of such a call to `checkMotionDone` similar to the remaining interface methods?

pattacini, 2015-02-02:
The same discussion applies to the method `checkMotionDone(bool *flag)` which in turn handles the whole set of joints.

pattacini, 2015-02-02:
Actually, I don't remember whether `checkMotionDone(bool *flag)` was returning one single bool already in the past. If it is so, then there is a coherence in keeping such a behavior also for the sub-set call.

barbalberto, 2015-02-03:
Yes, I copied the behaviour of the 'all joint' version while creating the 'group' one.

pattacini, 2015-02-03:
Then, for me there is a sort of continuity with the past and this feature-request can be withdrawn. I'll leave it to you guys.

lornat75, 2015-02-03:
There could be improvements, like:

1) defining a new checkMotionDone(bool &val) and deprecate checkMotionDone(bool *val) 
2) define a new checkMotionDone(int j, bool* vals) that works on multiple joints

All possible solutions, but I am not sure the problem is worth going through the interface redefinition...

pattacini, 2015-02-03:
I agree @lornat75 , no need to change anything, I was just caught trapped because of the method documentation. Solved this, I think it's perfectly fine.
Closing...

closed 2015-02-03 by pattacini