robotology/yarp #391: [yarpmanager]

alecive, 2015-02-03:
Currently, if I double-click on a port on `yarpmanager`, I can select the `inspect` item from the menu it appears, and I have these choices:
 * ``
 * ``

alecive, 2015-02-07:
Commit c52e110 has let me modify the `yarpmanger (QT)` in order to integrate the feature and start the datadumper with a  right click on the port to dump. Now the problem is related to the closure of the aforementioned datadumper: after starting, the `datadumper` runs in background,  and I honestly don't know what would be the best solution. Any idea would be welcome!

pattacini, 2015-02-07:
@alecive, you could send the "quit" request to the rpc port opened up by `yarpdatadumper`.

For example:
```
echo "quit" | yarp rpc /dump/rpc
```

closed 2015-02-10 by pattacini