robotology/yarp #392: How to use the yarpLogger

jeljaik, 2015-02-03:
I'm including `yarp/os/LogStream.h` in my code as done in https://github.com/robotology/icub-main/blob/c89def02c9094a0d008a1077e24521232e69c13b/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp allowing me to use the macros `yDebug`, `yError`, `yWarning` and so on. I was wondering what else is needed in order to have the yarpLogger properly working.

@randaz81

alecive, 2015-02-04:
@jeljaik as far as I understood, you have to run a `yarprun` with a `--log` argument, and run your sw with `yarpmanager` (or directly running it through a `yarprun`). Then, everything will automagically work :)

traversaro, 2015-02-04:
I have another question related to this: it is legal to do something like `yDebug() << std::endl` ? 
My code is failing if I try something like that, with the following (quite verbose) message:
```
/home/pegua/src/gazebo-yarp-plugins/src/yarp_drivers/ControlBoardDriver.cpp: In member function ‘bool yarp::dev::GazeboYarpControlBoardDriver::setJointNames()’:
/home/pegua/src/gazebo-yarp-plugins/src/yarp_drivers/ControlBoardDriver.cpp:242:79: error: no match for ‘operator<<’ (operand types are ‘yarp::os::LogStream’ and ‘<unresolved overloaded function type>’)
                      << " ( " << i << " of " << nr_of_joints << " ) " << "\n" << std::endl;
                                                                               ^
/home/pegua/src/gazebo-yarp-plugins/src/yarp_drivers/ControlBoardDriver.cpp:242:79: note: candidates are:
In file included from /home/pegua/src/gazebo-yarp-plugins/src/yarp_drivers/ControlBoardDriver.cpp:16:0:
/usr/local/include/yarp/os/LogStream.h:70:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(bool)
     inline LogStream& operator<<(bool t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:70:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘bool’
/usr/local/include/yarp/os/LogStream.h:75:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(char)
     inline LogStream& operator<<(char t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:75:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘char’
/usr/local/include/yarp/os/LogStream.h:80:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(short int)
     inline LogStream& operator<<(signed short t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:80:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘short int’
/usr/local/include/yarp/os/LogStream.h:85:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(short unsigned int)
     inline LogStream& operator<<(unsigned short t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:85:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘short unsigned int’
/usr/local/include/yarp/os/LogStream.h:90:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(int)
     inline LogStream& operator<<(signed int t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:90:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘int’
/usr/local/include/yarp/os/LogStream.h:95:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(unsigned int)
     inline LogStream& operator<<(unsigned int t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:95:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘unsigned int’
/usr/local/include/yarp/os/LogStream.h:100:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(long int)
     inline LogStream& operator<<(signed long t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:100:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘long int’
/usr/local/include/yarp/os/LogStream.h:105:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(long unsigned int)
     inline LogStream& operator<<(unsigned long t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:105:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘long unsigned int’
/usr/local/include/yarp/os/LogStream.h:110:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(long long int)
     inline LogStream& operator<<(signed long long t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:110:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘long long int’
/usr/local/include/yarp/os/LogStream.h:115:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(long long unsigned int)
     inline LogStream& operator<<(unsigned long long t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:115:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘long long unsigned int’
/usr/local/include/yarp/os/LogStream.h:120:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(float)
     inline LogStream& operator<<(float t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:120:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘float’
/usr/local/include/yarp/os/LogStream.h:125:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(double)
     inline LogStream& operator<<(double t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:125:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘double’
/usr/local/include/yarp/os/LogStream.h:130:23: note: yarp::os::LogStream& yarp::os::LogStream::operator<<(const char*)
     inline LogStream& operator<<(const char* t) {
                       ^
/usr/local/include/yarp/os/LogStream.h:130:23: note:   no known conversion for argument 1 from ‘<unresolved overloaded function type>’ to ‘const char*’
[... cut ...]
/home/pegua/src/gazebo-yarp-plugins/src/yarp_drivers/ControlBoardDriver.cpp:242:87: note:   ‘yarp::os::LogStream’ is not derived from ‘std::basic_ostream<_CharT, _Traits>’
                      << " ( " << i << " of " << nr_of_joints << " ) " << "\n" << std::endl;
                                                                                       ^
make[2]: *** [CMakeFiles/gazebo_yarp_controlboard.dir/src/yarp_drivers/ControlBoardDriver.cpp.o] Error 1
make[1]: *** [CMakeFiles/gazebo_yarp_controlboard.dir/all] Error 2
make: *** [all] Error 2
```

drdanz, 2015-02-04:
It is not legal, unless you want to define the operator<< for std::ostream...
You can use ``<< "\n"``

randaz81, 2015-02-04:
@traversaro It's also unnecessary because endlines are automatically added by the logging engine. Moreover, if multiple trailing '/n' are found, they are automatically removed, and just one is kept. So, even if you put a std::endl, it will be thrown away, unless you use it in the middle of the line.
In the same manner initial spaces in sentences like "     This is an message" are also automatically removed.

randaz81, 2015-02-04:
@jeljaik @alecive 
Even if launching a module with yarprun is generally preferred in our lab, you can use the logger also if you start the module from the command line. In this case, you just need to set the environment variable YARP_FORWARD_LOG_ENABLE=TRUE on your machine.

drdanz, 2015-02-04:
@randaz81 Are you sure? If I remember correctly, only the last ``\n`` is removed...

@jeljaik @alecive Please note that ``YARP_FORWARD_LOG_ENABLE`` is not documented and might be subject to change in the future. Also using it **with** yarprun is probably dangerous, or at least might cause the same messages to be sent twice.

francesco-romano, 2015-03-02:
@drdanz @randaz81 so what is the best way to run a module which has been migrated to use yarp_log and see the errors on the stdout / stderr?

alecive, 2015-07-20:
@jeljaik can we close this issue? 
@randaz81 @drdanz any update on this? A complete documentation on the logger and how it works would be nice.

vtikha, 2015-09-04:
Let's try this again shall we? @jeljaik @alecive @randaz81 @drdanz can we close this issue?

closed 2015-11-09 by jeljaik