robotology/yarp #393: problem to drag and close "yarpview --compact"

pattacini, 2015-02-03:
When the yarpview is launched with `--compact` option, the `Atl+Drag` feature is not available on Windows and is not even possible to be closed up.

pattacini, 2015-02-03:
Could we restore at least the `title bar`?

alecive, 2015-02-04:
We agreed with @pattacini to have two options from now on:

 * `--compact` will keep the titlebar but get rid of the statusbars
 * `--minimal` will remove the titlebar as well, for more extreme uses (and users)

alecive, 2015-02-04:
It should be fixed now.
@pattacini can you test the behavior of the new feature on windows?

pattacini, 2015-02-04:
it works.

alecive, 2015-02-05:
@pattacini what about putting the name of the yarpview port in the title bar instead of "YARP Qt image viewer" when in compact mode?

pattacini, 2015-02-05:
yep, it'd be a nice plus

commit dcdfbc41 (2015-02-04): Fixed #393: added --minimal alongside --compact argument. To be tested on windows
commit a2aa5e72 (2015-02-07): Fixed #393: Added name of the port to window title
closed 2015-02-07 by alecive