robotology/yarp #394: How to emulate yarp namespace and yarp detect from code

francesco-romano, 2015-02-04:
Hi.
I need to setup the yarp environment (so at least namespace and detect the server) directly from a .cpp application.
How can I do it?

This is what I'm trying:
```objective-c
    using namespace yarp::os;

    NSLog(@"Address %@", [self getIPAddress]);
    NSLog(@"Nameserver: %s", NetworkBase::getNameServerName().c_str());
    NetworkBase::setNameServerName("/fra");
    bool scanNeeded, serverUsed;
    Contact server = Network::detectNameServer(true, scanNeeded, serverUsed);

```

The log says:
~~~
2015-02-04 15:22:09.444 TestYarp[13788:256820] Address 192.168.36.75
2015-02-04 15:22:09.445 TestYarp[13788:256820] Nameserver: /fra
warning: YARP name server(s) not configured, ports will be anonymous
warning: check your namespace and settings with 'yarp detect'
yarp: No connection to nameserver
yarp: no connection to nameserver, scanning mcast
2015-02-04 15:22:24.745 TestYarp[13788:256925] Terminating since there is no system app.
~~~

~~~
yarp: Port /fra active at tcp://192.168.36.75:10000

Registering name server with itself:
 * register "/fra" tcp "192.168.36.75" 10000
   + set "/fra" ips "127.0.0.1" "192.168.36.75" "10.255.37.162"
   + set "/fra" process 13799
 * register fallback mcast "224.2.1.1" 10000
   + set fallback ips "127.0.0.1" "192.168.36.75" "10.255.37.162"
   + set fallback process 13799
Name server can be browsed at http://192.168.36.75:10000/

Ok.  Ready!
~~~

@paulfitz

jgvictores, 2015-02-05:
Hi Francesco, from http://robotcub-hackers.2198711.n2.nabble.com/Aquila-YARP-server-management-td7011435.html we have that Network::runNameServer(0,NULL) should work, but questions arise not only on "how to do it" but also regarding "should we do it?".
- "how to do it": I've found runNameServer to be blocking and haven't found a clean way to stop it when running in, say, a yarp::os::Thread. @paulfitz any ideas?
- "should we do it?": I understand it's generally not a good philosophy, but my use case is during unit testing (here's an example that needs serious refractoring: https://github.com/asrob-uc3m/robotDevastation/blob/master/test/testRdYarpImageManager.cpp). @francesco-romano which is your specific use case?

traversaro, 2015-02-05:
@jgvictores We (IIT) are in process of designing an flexible testing infrastructure for performing testing both as CI jobs (using simulated robots) and regression testing (using real robots). 
If you are interested, you can check this repo: https://github.com/robotology/robot-testing and its issues.

francesco-romano, 2015-02-05:
Maybe I wasn't completely clear.
I was not trying to launch a `yarpserver` from a c++ application.
I was trying to properly configure yarp inside a C++ application. 
My use case is the following:
I don't have access to a terminal to setup the usual
```bash
yarp namespace /mynamespace
yarp detect --write
```
and everything should  be self-contained in the application

paulfitz, 2015-02-05:
@francesco-romano do you know if your YARP is compiled with or without ACE? Your original test seems to work for me, with YARP compiled with ACE. Mcast hasn't been implemented for yarp without ace. 

Another option may be to use `Network::main(argc,argv)`, which you can just call with whatever you would pass to the `yarp` command-line. Watch out, the first element of argv should be filled with the program name (or anything you like, it will be ignored).

francesco-romano, 2015-02-05:
@paulfitz I'm trying a minimal yarp setup: 
```
cmake .. -GXcode -DSKIP_ACE:BOOL=TRUE -DCREATE_SHARED_LIBRARY:BOOL=FALSE -DYARP_USE_READLINE:BOOL=FALSE -DYARP_WRAP_STL_STRING:BOOL=FALSE
```

My idea is to make yarp running on iOS, so I don't have the command line :(

paulfitz, 2015-02-05:
@jgvictores what yarp does to have a nameserver during its own unit tests can be seen in https://github.com/robotology/yarp/blob/master/src/libYARP_OS/harness/harness.cpp (search for three instances of `YARP_USE_PERSISTENT_NAMESERVER`). This creates a nameserver that can be stopped at will. There's no public api for this though :(.

@francesco-romano ah I see. Without ACE, there's no multicast right now, which means automatic discovery of the name server won't work.  If you need automatic discovery, then either multicast needs to get added (may not be difficult) or another discovery method needed to be added.

francesco-romano, 2015-02-05:
@paulfitz I understand.
And.. if one can specify IP and port by hand? Can it work without ACE? If so, how can I then initialize it?

paulfitz, 2015-02-05:
@francesco-romano yes that doesn't need ace.  Here's an ugly way to do it:
```c++
  const char *argv[] = {
    "yarp",
    "conf",
    "192.168.1.99", // replace
    "10000" // replace
  };
  int result = NetworkBase::main(sizeof(argv)/sizeof(const char*),(char **)argv);
```
You could stick that between your `setNameServerName` and `detectNameServer`. The `scanNeeded` flag should not get set if the server is indeed found at the address you specify.

francesco-romano, 2015-02-05:
thanks @paulfitz. 
It seems to work.
Are there any clearer way to do this instead of "faking" a main?

paulfitz, 2015-02-05:
@francesco-romano `impl::NameConfig::setAddress(contact)` should do it, but we discourage references to implementation classes in user code. Maybe you could add a method in `NetworkBase` or `YarpNameSpace` that does the call?

paulfitz, 2015-02-10:
Shall we consider this question answered?

francesco-romano, 2015-02-11:
Sorry I forgot about it.

closed 2015-02-11 by francesco-romano