robotology/yarp #395: Standarize exit codes (mainly yarprun and RFModule)

jgvictores, 2015-02-05:
yarpserver returns a 1 upon failing to open, and a 130 on Control-C. That's great, because it complies with http://tldp.org/LDP/abs/html/exitcodes.html. My concerns:
- yarprun: returns a 255 upon failing to open (because -1 is out of range), and a 0 on Control-C. The failure can be confusing for a .profile script that launches yarprun (in absence of a tool like icub-cluster.py in yarp, and maybe even for a script like such).
- RFModule: returns a  255 upon failing to open (because -1 is out of range), and a 0 on Control-C. A 255 would be passed as a local exit code, however, yarprun attaches a -1 (YARPRUN_ERROR) to the response Bottle... some ambiguity here.
- yarpview: returns a 0 upon failing to open, a 130 on Control-C.

lornat75, 2015-02-05:
I am in favor of standardizing return values. My comments:

- ok for returning 1 on failure 
- 0 signal success I guess
- RFModule and yarprun catch Control-C and attempt smooth shutdown, in case of success I think they should return 0, 1 otherwise.

jgvictores, 2015-02-05:
@lornat75 Since bool return values within many functions limit the range to 2 different values, a 1 on failure and a 0 on success (obviating the originating signal) is a sound solution.

commit 9fa5c9c3 (2015-07-20): RFModule, yarpdev, yarprun, yarpserver, yarpdatadumper, yarpdataplayer, yarpmotorgui, yarpscope, yarpview return codes standardized: fixed #395
closed 2015-07-20 by pattacini