robotology/yarp #398: datadumper depends form opencv

lornat75, 2015-02-09:
This causes problems with the binary packages, at the moment YARP did not depend from opencv.

What are the functionalities that datadumper users from opencv? can we provide equivalent code?

lornat75, 2015-02-09:
@pattacini @drdanz @paulfitz

pattacini, 2015-02-09:
OpenCV is used to produce videos by means of `videoWriter` object. Providing equivalent code for that is quite tricky I guess.

Nonetheless, `yarpdatadumper` can be compiled withouth OpenCV allowing for bottles and images acquisition. Could we ship this version of the tool in yarp release? If one wants to have the video acquisition too, he needs to rely on yarp source.

lornat75, 2015-02-09:
It would be an solution.

pattacini, 2015-02-09:
Consider also that without OpenCV `yarpdataplayer` is not even compiled, I presume. Are we releasing `yarpdatadumper` without its player companion then? It becomes a bit weird, maybe.

vtikha, 2015-02-09:
We could consider removing any opencv dependencies from the yarpdataplayer. As of now it is only used to read and load the images saved from the yarpdatadumper. Opencv was chosen as it is able to handle multiple image formats while if I remember correctly yarp cannot at the moment. Typically for large datasets a ppm is converted to jpg or whatever to reduce space. But we can decide a standard way and make the changes accordingly.

paulfitz, 2015-02-09:
there's always the option of defining an interface for reading/writing files, then yarp could load an opencv plugin to do that on need. There's already image *stream* reader/writer interfaces.  Reading is implemented for opencv, but not yet writing. Both reading and writing are implemented with ffmpeg, but unmaintained. The ffmpeg_writer device used to work fine for writing videos, and I'd expect an opencv_writer device to be straightforward to write (and a lot easier to maintain).

lornat75, 2015-02-09:
I was not aware of datasetplayer. If opencv support can be disabled at compile time it would be ok I guess to come up with an initial binary release

lornat75, 2015-02-10:
@pattacini @vtikha @drdanz what do we need to do to implement this?

vtikha, 2015-02-10:
what do you mean with "this"? What @paulfitz suggested (which is a good idea) or what I mentioned to remove the OpenCV dependencies? If it is the second one, I can do it relatively soon.

lornat75, 2015-02-10:
This --> modify datadumper/datasetplayer so that they can be optionally compiled without opencv so we proceed with the binary release. Next we can implement Paul's idea.

pattacini, 2015-02-10:
`yarpdatadumper` already deals with that

drdanz, 2015-02-10:
I can confirm that:

  * ``yarpdatadumper`` is built *always* and has OpenCV as an optional dependency.
  * ``yarpdataplayer`` is built *only if OpenCV is found* (and obviously GTK/Qt), since it does not compile without it.

I don't know what it would take to make OpenCV since I didn't write the code, but I just took care of the build system...

vtikha, 2015-02-10:
Yes. It does not require much. I will take care of it

vtikha, 2015-02-10:
yarpdataplayer is now always built but has OpenCV as an optional dependency.

lornat75, 2015-02-10:
@mbrunettini

drdanz, 2015-02-11:
Backported all the changes to ``yarpdataplayer-gtk`` as well. Now all the ``yarpdataxxx` tools build with OpenCV as an optional dependency

closed 2015-02-11 by drdanz

vtikha, 2015-02-11:
Just small note...Shouldn't we take this opportunity to get rid of all the gtk versions? We have dedicated quite some time on debugging and testing of the various qt modules and it seems more than natural to remove them once a module is considered ready and functioning.

traversaro, 2015-02-11:
A lot of CI jobs for codyco-software are now failing while building master branch of yarp with this error (that I guess is related to this issue):
```
-- ending plugin library: yarpcar
-- Performing Test CXX_HAS_WNO_MISMATCHED_TAGS
-- Performing Test CXX_HAS_WNO_MISMATCHED_TAGS - Success
-- Not processing thrift file dataSetPlayer.thrift, ALLOW_IDL_GENERATION=OFF
CMake Error at src/yarpdataplayer-gtk/CMakeLists.txt:59 (target_link_libraries):
Cannot specify link libraries for target "yarpdataplayer-qt" which is not
built by this project.
-- Configuring incomplete, errors occurred!
See also "/home/travis/build/robotology/codyco-superbuild/build/external/YARP/CMakeFiles/CMakeOutput.log".
See also "/home/travis/build/robotology/codyco-superbuild/build/external/YARP/CMakeFiles/CMakeError.log".
make[2]: *** [external/YARP/CMakeFiles/YCMStamp/YARP-configure] Error 1
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
make: *** [all] Error 2
```

See for example: https://travis-ci.org/robotology/codyco-superbuild/jobs/50326298 

There is something wrong with the YARP options that we use (https://github.com/robotology/codyco-superbuild/blob/master/cmake/BuildYARP.cmake) ?

drdanz, 2015-02-11:
Fixed... sorry, copy & paste error :sweat:

drdanz, 2015-02-11:
I think we decided to keep the Qt versions until all the supported platforms have Qt5 (a few supported MSVC versions don't have it yet) and we have all guis ported to Qt (yarpbuilder and a few other guis in icub are still gtk only).
Therefore I believe we cannot get rid of GTK, but we should add new features to the Qt version only, but this was an exception since the patch was trivial and it would have required a ridicolous amount of `if` in CMake to support one version with OpenCV optional and the other one with OpenCV required.

vtikha, 2015-02-11:
My question was in general and not related to the patch. Adding new features only to Qt is fine, but I do not see why we we need to wait to have all other guis ported in Qt in order to remove the gtk versions. We can gradually remove them and clean the repo.

alecive, 2015-02-11:
I agree with @vtikha :+1:

lornat75, 2015-02-11:
I think the idea was to keep both until the qt versions were stable. Binaries package will need to provide gtk until there are gtk guis in icub.

About qt5 and MSVC, I think we can move forward the supported visual studio compilers to Visual Studio 10/2010 (for which there is qt5) and subsequent.

All in all this will allow us to remove gtk guis that are already re-implemented in qt5.