robotology/yarp #402: Carriers.h should be moved out from impl

drdanz, 2015-02-11:
Any 3rd party project using yarp should not use impl files, but in order to create carrier plugins, the yarp::os::impl::Carriers::addCarrierPrototype() is called by the template.

We should move ``yarp/os/impl/Carriers.h`` into ``yarp/os/Carriers.h`` and the class ``yarp::os::impl::Carriers`` to ``yarp::os::Carriers`` in order to solve this issue.

It shouldn't be hard to do, the only requirements is to use PIMPL to hide the usage of ``yarp::os::impl::PlatformVector<Carrier *>``.

paulfitz, 2015-02-11:
or expose `addCarrierPrototype` elsewhere in the api, e.g. as a static method of `yarp::os::Carrier`?

drdanz, 2015-02-11:
Both work for me, even if the I was suggesting to move it in analogy to ``yarp/dev/Drivers.h``

PR #877 (2016-08-19): YARP_OS: Move Carriers out of the impl namespace
Fixes #402

This will allow to create carriers as plugins in separate packages.
<a href='#crh-start'></a><a href='#crh-data-%7B%7D'></a>
[image]
[image]&nbsp;[image]
<a href='#crh-end'></a>
commit 2eb4bf6e (2016-08-19): YARP_OS: Move Carriers out of the impl namespace

Fixes #402
closed 2016-08-19 by drdanz