robotology/yarp #408: yarpdev OpenNI2Device takes ages to start

kt10aan, 2015-02-13:
Hi,

I updated yarp today after some time (last update was sometime in December 2014).

I wanted to test the OpenNI2DeviceServer device for PR#407

In any case, I observed that when I issue the yarpdev --device OpenNI2DeviceServer command, the driver starts immediately with the messages about opening the ports, then waits forever (~1 minute!) until the main part of the driver is loading (the one communicating with the sensor etc). Before, it was almost instantaneous!

Is it the result of recent changes to yarp?

Should I look somewhere else for the cause? The thing is that the non-yarp software (OpenNI2 and NiTE2) have not changed since the last time.

The output is (with a comment added to show where the delay occurs):
```
mini:~ edmund$ yarpdev --device OpenNI2DeviceServer --noRGB --noUserTracking --printVideoModes
Starting OpenNI2 YARP Device please wait...
yarp: Port /OpenNI2:i active at tcp://192.168.0.14:10002
yarp: Port /OpenNI2/depthFrame:o active at tcp://192.168.0.14:10004
//-------------------
// This is where the driver is waiting forever.....
//---------------------
Device Name: PS1080
OpenNI v2.2.0.33
NiTE v2.2.0.11
Image registration mode is supported
Image registration mode is off
Depth/Color frame sync disabled
Depth video mode: 0 - 320x240 - 30 fps
Depth stream started...
Depth video modes available:
0: 320x240, 30 fps
2: 320x240, 60 fps
4: 640x480, 30 fps
6: 160x120, 30 fps
OpenNI2 Yarp Device started.
[INFO]created device <OpenNI2DeviceServer>. See C++ class  for documentation.
yarp: Port /OpenNI2DeviceServer/quit active at tcp://192.168.0.14:10006
[INFO]device active...
```

paulfitz, 2015-02-13:
hmm, no cause leaping to mind... skimming the code, I don't see much yarpy stuff going on in that interval, but I could certainly be missing something...

kt10aan, 2015-02-13:
Well, some more information. It seems now that the delay is not only happening where I mentioned above but also at the point where "OpenNI2YarpDevice started." comes up (I was just focusing on the first delay before).

so:
```
mini:~ edmund$ yarpdev --device OpenNI2DeviceServer --noUserTracking
Starting OpenNI2 YARP Device please wait...
yarp: Port /OpenNI2:i active at tcp://192.168.0.14:10002
yarp: Port /OpenNI2/depthFrame:o active at tcp://192.168.0.14:10003
yarp: Port /OpenNI2/imageFrame:o active at tcp://192.168.0.14:10005
//------------------------
// First Delay
//----------------------
Device Name: PS1080
OpenNI v2.2.0.33
NiTE v2.2.0.11
Image registration mode is supported
Image registration mode is off
Depth/Color frame sync disabled
Depth video mode: 0 - 320x240 - 30 fps
Depth stream started...
RGB video mode: 0 - 320x240 - 30 fps
RGB stream started...
//-----------------------------------------------------
// Second delay
//------------------------------------------------------
OpenNI2 Yarp Device started.
[INFO]created device <OpenNI2DeviceServer>. See C++ class  for documentation.
yarp: Port /OpenNI2DeviceServer/quit active at tcp://192.168.0.14:10004
[INFO]device active...
[INFO]device active...
[INFO]device active...
[INFO]device active...
[INFO]device active...
[INFO]device active...
^C[INFO][try 1 of 3] Trying to shut down /OpenNI2DeviceServer/quit
Destroying depth stream...Done
Destroying RGB stream...Done
Closing sensor device...Done
Closing ports...Done

Again, I should mention with no change at the driver itself, or any of the other software. The only thing changed was YARP. And the delays *seem* to happen (i make no claims about deep knowledge of the plugin system) when there is the exchange back and forth from the yarp driver scaffolding to my code. (I know its an awful description :D, sorry!)

paulfitz, 2015-02-13:
@kt10aan do you have a stack trace or function names when the delay is happening? Not set up to run the device locally :-(

gsaponaro, 2015-02-14:
I will test this next week, with an old and a new yarp.

kt10aan, 2015-02-15:
The original report was done on a mac.

Well, I tested it on an Ubuntu 14.04 machine that had an older YARP. Then I updated YARP and tested it again. On Ubuntu (and I assume linux in general) the driver is fine and starts up normally (@gsaponaro, no need for testing it :)

@paulfitz, I will get a stack trace from the mac, first I have to check the manual of lldb :D

kt10aan, 2015-02-15:
Aaaaand, as on Ubuntu everything was working fine, I tried to eliminate other sources of problems...
It turns out that for some reason, removing an SD card from the machine eliminated the issue...Yes I am baffled, I can only assume that the SD card reader was on the same usb channel as the Xtion or something else obscure?
Well, I still have to learn lldb commands :D

@gsaponaro, maybe it is something to go in the "Troubleshooting" section of the OpenNI2 driver documentation?

@paulfitz , sorry for the wasted time, close the issue please :)

paulfitz, 2015-02-15:
no problem, feel free to reopen if the trail eventually leads to yarp

closed 2015-02-15 by paulfitz