robotology/yarp #412: yarpdataplayer has references to old name dataSetPlayer

gsaponaro, 2015-02-17:
```yarpdataplayer``` (Qt version) has a bunch of references to its previous name dataSetPlayer, which should probably be updated to avoid confusion:

* GUI window name
* default port name ```/dataSetPlayer/rpc:i```
* in the sources, not directly affecting the user, just the developers: thrift file, IDL generated files, default context set in main.cpp, QtDataSetPlayer.pro (QtCreator file that can likely be deleted from the repository)

I am using yarp 91b4edf43c21aa6a09f4095504c5ab917dff7708

vtikha, 2015-02-18:
Hi @gsaponaro I am currently doing the changes. But  the GUI window name and the default port name were changed when the QT version was committed. In any case I am going through it all in the CMakeFiles and IDL. I am not sure about what to do with the .pro but I will ask and test before committing anything.

gsaponaro, 2015-02-18:
Regarding the first two points of my list: strike them, you're right. I thought that deleting CMake cache and recompiling yarp was a sufficient test before I filed this issue, but I had to do so for icub-main too (I had stale references to dataSetPlayer lying around in icub-main/build/share)... sorry.

Regarding the last point, in case you want to remove all references to the old name in yarpdataplayer-qt, ```grep -R dataSetPlayer``` gives a few.

vtikha, 2015-02-18:
The module yarpdataplayer has been restructured and all references to the old dataSetPlayer have been removed. Everything has been committed and you do not need to generate the IDL files again.

gsaponaro, 2015-02-18:
Cool.

gsaponaro, 2015-09-16:
This page mentions the old name of this module:
http://wiki.icub.org/yarpdoc/check_your_installation.html

vtikha, 2015-09-16:
Thanks @gsaponaro for notifying, changes have been committed.

closed 2015-09-16 by vtikha