robotology/yarp #421: YARP in OS X generates lots of warning after ACE upgrade

kt10aan, 2015-02-23:
Well, technically it is only two warnings but they happen in many many places, resulting in way too much noise.

Both have to do with deprecated functions. 
They are not affecting anything for now but I assume that these warnings will come to Linux eventually.

```
In file included from /usr/local/Cellar/ace/6.3.0/include/ace/OS_NS_Thread.h:2064:
/usr/local/Cellar/ace/6.3.0/include/ace/OS_NS_Thread.inl:1652:28: warning:
      'sem_init' is deprecated [-Wdeprecated-declarations]
      ACE_OSCALL_RETURN (::sem_init (s->sema_,
                           ^
/usr/local/Cellar/ace/6.3.0/include/ace/config-macros.h:312:20: note:
      expanded from macro 'ACE_OSCALL_RETURN'
    return (TYPE) (X); \
                   ^
/usr/include/sys/semaphore.h:55:5: note: 'sem_init' has been explicitly
      marked deprecated here
int sem_init(sem_t *, int, unsigned int) __deprecated;
```

and the second one:
```
In file included from /usr/local/Cellar/ace/6.3.0/include/ace/OS_NS_Thread.h:2064:
/usr/local/Cellar/ace/6.3.0/include/ace/OS_NS_Thread.inl:1453:21: warning:
      'sem_destroy' is deprecated [-Wdeprecated-declarations]
      ACE_OSCALL (::sem_destroy (s->sema_), int, -1, result);
                    ^
/usr/local/Cellar/ace/6.3.0/include/ace/config-macros.h:316:22: note:
      expanded from macro 'ACE_OSCALL'
    RESULT = (TYPE) (X); \
                     ^
/usr/include/sys/semaphore.h:53:5: note: 'sem_destroy' has been explicitly
      marked deprecated here
int sem_destroy(sem_t *) __deprecated;
```

paulfitz, 2015-02-23:
Not totally sure we can do something about this in yarp. On OSX yarp uses native semaphores, not stuff from ACE. This looks at first glance like a warning triggered by ACE itself, in `ace/OS_NS_Thread.h`. I don't think we can avoid (indirectly) including that file. There might be a way to tweak the homebrew ACE formula that controls how ACE is being configured.

Depending on your use case, one other option is to compile yarp without ace. Most things will work. Biggest thing not yet reimplemented is multicast support.

francesco-romano, 2015-02-23:
Same warnings as in https://github.com/robotology/yarp/issues/148

francesco-romano, 2015-02-23:
Maybe I'm wrong, but taking a look at https://github.com/DOCGroup/ATCD/blob/master/ACE/ace/OS_NS_Thread.inl
it seems to me that on OS X the macro `ACE_LACKS_NAMED_POSIX_SEM` is defined while it should not (on OS X unnamed semaphore are deprecated, and instead we should use the named version)

paulfitz, 2015-02-23:
that sounds like the kind of thing that could fix this. One way to test is to modify https://github.com/Homebrew/homebrew/blob/master/Library/Formula/ace.rb locally to tweak the config.h and see if you can get something happier. The homebrew people have in the past accepted little kludges in the formula as long as we made the effort to get the fix upstream to ace so that the kludge could eventually be dropped.

francesco-romano, 2016-08-05:
I'm not experiencing anymore this warning.
I close this issue.
Feel free to open it again if the issue is still present

closed 2016-08-05 by francesco-romano