robotology/yarp #424: Documentation on Qt5 dependencies on Ubuntu 14.04 is not correct

kt10aan, 2015-02-26:
The Qt5 dependencies for Ubuntu Trusty listed at http://wiki.icub.org/yarpdoc/dependencies.html, are not sufficient.
With the above dependencies installed, I get the following error when running yarpview:
```
QQmlApplicationsEngine failed to load component
qrc:/qml/QtYARPView/main.qml:52 Type VideoSurface unavailable
qrc:/YARPView/VideoSurface.qml:14 module "QtQuick.Dialogs" is not installed
```

Since that was my first installation of Trusty that I used the ubuntu packages and not the Qt5 installer, I cannot seem to find which package is missing.

drdanz, 2015-02-26:
@kt10aan The package should be qtdeclarative5-dialogs-plugin or something similar, can you check if it exists and if it fixes your issue?

kt10aan, 2015-02-26:
@drdanz , thanks! Installing qtdeclarative5-dialogs-plugin fixed the issue :)

drdanz, 2015-02-26:
@kt10aan Updated docs (will be online as soon as the documentation is re-generated), thanks for reporting.

commit 4b489fd7 (2015-02-26): [docs] Fix Qt5 dependencies for Ubuntu trusty

Fixes: #424
closed 2015-02-26 by drdanz