robotology/yarp #426: portmonitor carrier does not work on my YARP installation

Ommac, 2015-03-02:
Hi,

I need to use portmonitor in a YARP application of mine, which has to run on the robot. It used to work some months ago, but now something is not working with my installation of YARP.

In fact, any connection using the portmonitor carrier produces the following message on the receiver side:

```
Carrier "portmonitor" does not modify incoming data as expected.
yarp: Removing input from /write to /read
```

-  I am using YARP on Ubuntu 14.04
-  I have followed the steps provided in http://wiki.icub.org/yarpdoc/portmonitor.html#need
-  I have recompiled and built the latest version of YARP.
-  I have used this example as a test: http://wiki.icub.org/yarpdoc/simple_modification.html and it still provides the same message, so the issue does not seem to depend on my application
-  I have tried to set the LUA_CPATH env var in the ~/.profile (then I issued the "source .profile" command) in the following ways, with no success:
  -  export LUA_CPATH=";;;$YARP_ROOT/bindings/build/?.so"
  -  export LUA_CPATH=";;;$YARP_ROOT/bindings/build/yarp.so"
  -  export LUA_CPATH=";;;$YARP_ROOT/bindings/build/lib/lua/?.so"

Does anybody see the problem here?
May I provide additional information?

Thank you,
Raffaello

traversaro, 2015-04-01:
I can confirm I have the same problem on Ubuntu 14.04 , using the example https://github.com/robotology/yarp/tree/master/example/portmonitor/coder_decoder . As an additional info, in my system yarp-lua bindings are correctly working , because I am testing several lua-based coordination components without problems.

drdanz, 2015-04-01:
@apaikan any idea?

apaikan, 2015-04-01:
A quick test has shown that the `modifiesIncomingData` method of `ModifyingCarrier` class from which the portmonitor modifier class is inherited is not called! 

There has not been any modification to the `portmonitor` carrier for months. So we need to see which commit have broken the carrier... 

I am guessing the problem probably should happen with any carrier modifiers in YARP.

Ommac, 2015-04-01:
Note: I haven't found any solution to this issue in the meantime.

apaikan, 2015-04-01:
finally! 

The commit 7b8a25f48b534f18fa500d61b0d9d29d610284bc has broken the `portmonitor` plugin: 
This has been removed from the `srx/carriers/CMakeLists.txt` : 
`set_property(GLOBAL APPEND PROPERTY YARP_LIBS yarpcar ${yarpcar_LIBRARIES})`

drdanz, 2015-04-01:
Should be fixed now, @Ommac, @apaikan can you confirm please?

closed 2015-04-01 by drdanz

traversaro, 2015-04-02:
I can confirm that is now working.

apaikan, 2015-04-02:
me too. good! 

lets close the issue!

commit 8d21b609 (2015-04-01): Add missing functions to ForwardingCarrier

Fixes #426