robotology/yarp #427: 2 controlBoardWrappers and 1 remoteControlBoard

barbalberto, 2015-03-03:
We still are in the situation in which we have 2 controlBoardWrappers and 1 remoteControlBoard.
This keeps creating new issues of compatibility, the last one shown up adding the protocol version check.
Also moving the setPositionMove from rpc to streaming is an issue, I'll update also the old one.
There are already a bunch of workaround to stretching the compatibility but I think it can only go worse as time goes by.
Why don't we simply freeze the version of the remoteControlBoard (renaming it) to be compliant with the old controlBoard, so that we have the devices coupled and fix the compatibility problem once and for all?

@lornat75 @paulfitz @drdanz ?

lornat75, 2015-03-03:
Fine by me.

traversaro, 2015-09-02:
Fixed by https://github.com/robotology/yarp/pull/493 .

closed 2015-09-02 by traversaro