robotology/yarp #430: Conversion encodes-ticks/degrees in IMotor

pattacini, 2015-03-05:
This is a simple reminder to extend in the future the new `yarp::dev::IMotor` interface in order to retrieve online the conversion factors between encoders ticks and degrees. This is especially useful for modules performing online tuning of the controllers gains.

pattacini, 2015-03-16:
Further, we could also implement a way to provide high-level modules other information stored in the xml files, such as mode of PID gains (machine or physical units)

pattacini, 2016-03-31:
@randaz81 as discussed, we should revive this issue. The reading of **encoders ratio** is now implemented for `ETH` robots but is missing for `CAN` robots. As agreed, we should finish this job when you'll have free time for this development :smirk: 

Also, what about mode of PID gains?

traversaro, 2016-03-31:
fyi @fjandrad