robotology/yarp #431: Adding "true" and "false" strings to Bottle produces unwanted quotation marks

gsaponaro, 2015-03-05:
I have a module that needs to add the words true/false (actual strings, not boolean values) to a Bottle. However I observed that adding the strings ```true``` and ```false``` to a Bottle then printing it on screen shows quotation marks around these words, while other strings show no quotes. I guess it has to do with true and false being reserved keywords in C++.

Example:

    Bottle bottle;
    bottle.addString("word1"); 
    bottle.addString("true");
    bottle.addString("any_other_word");
    cout << bottle.toString().c_str() << endl;

prints ```word1 "true" any_other_word``` <-- notice quotation marks around ```true```. Is it possible to obtain ```word1 true any_other_word``` instead?

gsaponaro, 2015-03-11:
In the end, this is not really a problem for our project: the client that reads the Values from my Bottle works whether those quotation marks are present or not.

However, I am still curious about this behaviour, if is is expected of yarp to put those quotes around true/false, do you know @paulfitz @drdanz?

Please feel free to close this issue when you want.

paulfitz, 2015-03-12:
I think this behavior is to keep unquoted true/false free for eventual json compatibility. When converting the entire bottle, each element is serialized in a way that lets it be read again as the original type.  You could ask `bot.get(1).asString()` if you expect a string, and there will be no quoting. Sorry for the confusion and delay responding.

gsaponaro, 2015-03-12:
Thank you, no problem.

/cc #351

closed 2015-05-04 by gsaponaro