robotology/yarp #433: error compiling YARP python bindings

arocchi, 2015-03-06:
While compiling yarp python bindings, I get this error:

```
/home/arocchi/walkman/build/external/YARP/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_new_StubDriverCreator(PyObject*, PyObject*)’:
/home/arocchi/walkman/build/external/YARP/bindings/yarpPYTHON_wrap.cxx:56565:154: error: no matching function for call to ‘yarp::dev::StubDriverCreator::StubDriverCreator(const char*, const char*, const char*, const char*)’
     result = (yarp::dev::StubDriverCreator *)new yarp::dev::StubDriverCreator((char const *)arg1,(char const *)arg2,(char const *)arg3,(char const *)arg4);
                                                                                                                                                          ^
/home/arocchi/walkman/build/external/YARP/bindings/yarpPYTHON_wrap.cxx:56565:154: note: candidates are:
In file included from /home/arocchi/walkman/external/YARP/src/libYARP_dev/include/yarp/dev/PolyDriver.h:13:0,
                 from /home/arocchi/walkman/external/YARP/src/libYARP_dev/include/yarp/dev/DriverLinkCreator.h:13,
                 from /home/arocchi/walkman/external/YARP/src/libYARP_dev/include/yarp/dev/all.h:22,
                 from /home/arocchi/walkman/build/external/YARP/bindings/yarpPYTHON_wrap.cxx:4131:
/home/arocchi/walkman/external/YARP/src/libYARP_dev/include/yarp/dev/Drivers.h:148:5: note: yarp::dev::StubDriverCreator::StubDriverCreator(const char*, const char*, const char*, const char*, const char*)
     StubDriverCreator(const char *name, const char *wrap, const char *code,
     ^
/home/arocchi/walkman/external/YARP/src/libYARP_dev/include/yarp/dev/Drivers.h:148:5: note:   candidate expects 5 arguments, 4 provided
/home/arocchi/walkman/external/YARP/src/libYARP_dev/include/yarp/dev/Drivers.h:144:31: note: yarp::dev::StubDriverCreator::StubDriverCreator(const yarp::dev::StubDriverCreator&)
 class YARP_dev_API yarp::dev::StubDriverCreator : public DriverCreator {
                               ^
/home/arocchi/walkman/external/YARP/src/libYARP_dev/include/yarp/dev/Drivers.h:144:31: note:   candidate expects 1 argument, 4 provided
make[6]: *** [bindings/CMakeFiles/_yarp_python.dir/yarpPYTHON_wrap.cxx.o] Error 1
```

arocchi, 2015-03-25:
Problem still present in current master.
Any ideas?

paulfitz, 2015-03-26:
@arocchi are you confident that the file `yarpPYTHON_wrap.cxx` is actually getting rewritten when the bindings are generated? Or could it be generated from a different version of Yarp lurking somewhere?  It looks inconsistent with the header files. Building the bindings locally, I get:
```c++
result = (yarp::dev::StubDriverCreator *)new yarp::dev::StubDriverCreator((char const *)arg1,(char const *)arg2,(char const *)arg3,(char const *)arg4,(char const *)arg5);
```
which has the correct number of arguments (5 rather than 4) and compiles well.

arocchi, 2015-03-26:
Indeed, my bad. Compiles succesfully now. Closing issue

closed 2015-03-26 by arocchi