robotology/yarp #435: [Plugins] Unexpected behaviour when DeviceDriver/Carrier is not the first inherited class

drdanz, 2015-03-09:
When deleting a device/carrier from a plugin, if the `yarp::dev::DeviceDriver`/`yarp::os::Carrier` class is not the first inherited class, something completely unexpected can happen (I noticed this because instead of the destructor of the class a complete different function was called by `delete`.
From the `YARP_DEFINE_SHARED_SUBCLASS` definition:

```c++
#define YARP_DEFINE_SHARED_SUBCLASS(factoryname,classname,basename) \
    YARP_SHARED_CLASS_FN void *factoryname ## _create () { return (basename *)new classname; } \
    YARP_SHARED_CLASS_FN void factoryname ## _destroy (void *obj) { delete (classname *)obj; } \
// [...]
```

If the class is not the first, then ``return (basename *)new classname;`` will return a different address from the one returned by `new` (i.e. "`classname *` + `offset`") .
Then the object is used, but when it is destroyed it is casted to `classname *` from a `void *`, therefore not using any knowledge about the base class. Therefore the pointer to "`classname *` + `offset`" is interpreted as if it was just "`classname *`", and instead of the destructor (that should be at "`classname *` + `delete-offset`") some random call is executed (the one found at "`classname *` + `offset` + `delete-offset`".

Depending on what is called, this can break horribly everything and debugging can be very painful.

In order to fix this, it is possible to double-cast the pointer to `basename *` first and to `classname *` after:

```c++
    YARP_SHARED_CLASS_FN void factoryname ## _destroy (void *obj) { delete (classname *)(basename *)obj; } \
```

casting ``void *`` to ``basename *`` and then to ``classname *`` will automatically take into account the `offset` and everything should work.

@paulfitz I could just change that line, but I have a couple of issues about this:

* Which kind of casts should we use? `static_cast`/c-style cast is probably the right thing to do from `void *` to `basename *`. For the cast from `basename *` to `classname *` both `dynamic_cast` and `static_cast` will work, but I'm not sure if conceptually `static_cast` is correct.
* Even though we just bumped the plugin systemVersion number, I'd like to bump it again, to ensure that we don't have broken plugins around, is it ok for you?

```c++
    YARP_SHARED_CLASS_FN void factoryname ## _destroy (void *obj) { delete (classname *)(basename *)obj; } \
```

paulfitz, 2015-03-09:
@drdanz yikes, good catch. Yes, please do bump the plugin version number. I defer to your judgement on the casts.

drdanz, 2015-03-09:
@paulfitz Does this look good to you?

```c++
#define YARP_DEFINE_SHARED_SUBCLASS(factoryname,classname,basename) \
    YARP_SHARED_CLASS_FN void *factoryname ## _create () { return static_cast<void *>(dynamic_cast<basename *>(new classname)); } \
    YARP_SHARED_CLASS_FN void factoryname ## _destroy (void *obj) { dynamic_cast<classname *>(static_cast<basename *>(obj)); } \
```

I'm changing both `create` and `destroy` to make both the casts explicitly, so that it is a bit easier to understand what is happening here.

drdanz, 2015-03-10:
After some more thinking, I'd change it in way that both `_create()` and  `_destroy()` are safe, and both casts explicit, so that it is obvious what it is happening.

```c++
#define YARP_DEFINE_SHARED_SUBCLASS(factoryname,classname,basename) \
    YARP_SHARED_CLASS_FN void *factoryname ## _create () { \
        classname *cn = new classname; \
        basename *bn = dynamic_cast<basename *>(cn); \
        if (!bn) delete cn; \
        return static_cast<void *>(bn); \
    } \
    YARP_SHARED_CLASS_FN void factoryname ## _destroy (void *obj) { delete dynamic_cast<classname *>(static_cast<basename *>(obj)); } \
```

* `_create()` will return a valid `basename` or a null pointer if `classname` does not inherit from `basename`.
* `_destroy()` will ensure that we are calling `classname` destructor even if `basename` is not the first inherited class. If the dynamic_cast fails, it will not delete the object (that is probably leaked), but it is less dangerous than just executing some other random function.

@paulfitz Does it look ok, or do you think that it is too much overhead?

PR #436 (2015-03-13): [YARP_OS] Properly cast the object in plugin create and destroy functions
When deleting a device/carrier from a plugin, if the
yarp::dev::DeviceDriver/yarp::os::Carrier class is not the first
inherited class, instead of the destructor of the class a complete
different function was called by delete.

The double cast in the _create() and _destroy() functions are required
to ensure that everything works when `basename` is not the first
inherited class:
 *  _create() will return a valid `basename` or a null pointer if
   `classname` does not inherit from `basename`.
 * _destroy() will ensure that  [... cut]
commit c114635c (2015-03-13): [YARP_OS] Properly cast the object in plugin create and destroy functions

When deleting a device/carrier from a plugin, if the
yarp::dev::DeviceDriver/yarp::os::Carrier class is not the first
inherited class, instead of the destructor of the class a complete
different function was called by delete.

The double cast in the _create() and _destroy() functions are required
to ensure that everything works when `basename` is not the first
inherited class:
 *  _create() will return a valid `basename` or a null pointer if
   `classname` does not inherit from `basename`.
 * _destroy() will ensure that [... cut]
closed 2015-03-13 by paulfitz