robotology/yarp #439: Portaudio

barbalberto, 2015-03-12:
A strange behaviour has been reported  by @rea regarding the portaudio device, I'll try to summarize.
Tests were done using portaudio device from the pc104 to record from the iCub microphones and played back on a consol pc with speakers.
Using the following command, reported in the page http://eris.liralab.it/wiki/Grabbing_sound_with_PortAudio

  yarpdev --device portaudio --id 1 --c 2 --read --rate 48000
The device start, but immediatly report "buffer underrrun error" and does not acquire the sound from the microphones. If we use yarphead to record to file, it'll be empty.
This command was the main way experiments were set-up and it was working up to few months ago, and it would be nice to have it working again.

If we use the examples files in yarp/example/portaudio, in particular with
sound_sender_mic on the pc104
sound_receiver on the pc with speakers
we are able to hear back from the speakers what the microphone is hearing but we saw other 2 strange facts.
- every second or so we see a print telling the sound_sender is resetting and from the speakers we get a bit of silence (white noise, then no white noise for short time, then white noise starts again).
- using yarphear to record this stream, the file recorded is a .wav but it is not playable with any common sound player.

Just for testing, I went back with yarp repo to the last commit done on portaudio library, it was 04/2014, but the behaviour did not change.
Portaudio is loaded as a plugin, but I don't think this could create such a issue, problem #423 was fixed by @randaz81 and it does not crashes anymore.

Can someone help debugging this strange behaviour?
@randaz81 @lornat75 @drdanz @paulfitz

AGiagkos, 2015-10-05:
Hello all,

We have noticed exactly the same issues when trying yarpdev/yarphear (buffer underrun error) and sound_sender_mic/sound_receiver. For the latter, on pc104 the sound_sender_mic outputs  "acquired xx seconds" and seems to work, on the desktop's side the sound_receiver outputs:

Device number 10
***** driver configuration changed, resetting *****
Device number 10
***** driver configuration changes, resetting *****

.. etc every ~1 second and the sound we get is very noisy.

Any comments on this are very welcome,

Many thanks, 
Alexandros

TataLab, 2015-10-21:
Hi Alexandros,

Francesco Rea and I have been working quite a lot with iCub audio.  We've developed a newer interface that seems stable and accurate.  It streams audio from PC104 over a YARP port.  We're also using some MATLAB .mex code to connect to that port and grab audio into the MATLAB workspace.  It's stable and working but we need to clean it and re-check before sharing it.  After that we'd be happy to share this interface for you to try.

Best Regards,

Matt Tata

AGiagkos, 2015-10-22:
Hello Matt,

Thanks you for your comment. That is great news, I'm looking forward to seeing your interface!

Alexandros

reafrancesco, 2016-06-04:
@AGiagkos 
I can now give you further details about the draft version  of the mentioned audio interface (available in robotology).
Please do not hesitate to contact me if you are still working on this.
Best,
Francesco