robotology/yarp #440: yarpmotorgui-qt crashes when no robot is running

pattacini, 2015-03-12:
Verified on Windows.
Further, the tool does not seem to check whether `yarpserver` is running. It proceeds to try to start up anyway.

alecive, 2015-03-12:
It happens also on linux. Here are my two tests:

1. without even the `yarpserver` : 
```
alecive@malakim:~/Programmazione/iCub/icub-main/build (master)$ yarpmotorgui
||| configuring
||| no policy found
||| default config file specified as yarpmotorgui.ini
||| checking [/home/alecive/Programmazione/iCub/icub-main/build/yarpmotorgui.ini] (pwd)
||| checking [/home/alecive/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/alecive/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| found /home/alecive/.local/share/yarp/robots/default
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/ICUBcontrib/robots/default] (robot yarp.d)
||| checking [/usr/local/share/iCub/robots/default] (robot yarp.d)
||| checking [/home/alecive/.local/share/yarp/robots/default/yarpmotorgui.ini] (robot)
||| checking [/home/alecive/.config/yarp/yarpmotorgui.ini] (YARP_CONFIG_HOME)
||| checking [/home/alecive/.local/share/yarp/yarpmotorgui.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/ICUBcontrib/yarpmotorgui.ini] (yarp.d)
||| checking [/usr/local/share/iCub/yarpmotorgui.ini] (yarp.d)
||| did not find yarpmotorgui.ini
yarp: cannot read from yarpmotorgui.ini
icub
Setting default parts.
APPENDING  "head" 
APPENDING  "torso" 
APPENDING  "left_arm" 
APPENDING  "right_arm" 
APPENDING  "left_leg" 
APPENDING  "right_leg" 
Checking the existence of: /icub/yarpmotorgui0/head/rpc:o 
yarp: No connection to nameserver
yarp: *** try running: yarp detect ***
ADDRESS is:  
yarp: Cannot connect to port /icub01
yarp: Port /icub/yarpmotorgui0/head/rpc:o failed to activate (invalid address)
yarp: Cannot connect to port /icub01
yarp: Port /icub/yarpmotorgui0/head/command:o failed to activate (invalid address)
yarp: Cannot connect to port /icub01
yarp: Port /icub/yarpmotorgui0/head/state:i failed to activate (invalid address)
yarp: Cannot connect to port /icub01
yarp: Port /icub/yarpmotorgui0/head/stateExt:i failed to activate (invalid address)
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
Setting a valid finder 
yarp: Cannot connect to port /icub01
yarp: Port /yarpmotorgui/head/sequence:o failed to activate (invalid address)
Opening interfaces...Segmentation fault (core dumped)
```

2. without the iCub_SIM
```
alecive@malakim:~/Programmazione/iCub/icub-main/build (master)$ yarpmotorgui
||| configuring
[... 32 lines cut ...]
yarp: cannot read from yarpmotorgui.ini
icub
Setting default parts.
APPENDING  "head" 
APPENDING  "torso" 
APPENDING  "left_arm" 
APPENDING  "right_arm" 
APPENDING  "left_leg" 
APPENDING  "right_leg" 
Checking the existence of: /icub/yarpmotorgui0/head/rpc:o 
ADDRESS is:  
yarp: Port /icub/yarpmotorgui0/head/rpc:o active at tcp://10.255.10.167:10105
yarp: Port /icub/yarpmotorgui0/head/command:o active at tcp://10.255.10.167:10106
yarp: Port /icub/yarpmotorgui0/head/state:i active at tcp://10.255.10.167:10107
yarp: Port /icub/yarpmotorgui0/head/stateExt:i active at tcp://10.255.10.167:10108
Problem connecting to /icub/head/rpc:i, is the remote device available?
Problem connecting to /icub/head/command:i, is the remote device available?
Problem connecting to /icub/head/state:o from /icub/yarpmotorgui0/head/state:i, is the remote device available?
*** Extended port /icub/head/stateExt:o was not found on the controlBoardWrapper I'm connecting to. Falling back to compatibility beaviour
Updating to newer yarp and the usage of controlBoardWrapper2 is suggested***
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
Setting a valid finder 
yarp: Port /yarpmotorgui/head/sequence:o active at tcp://10.255.10.167:10048
Opening interfaces...Segmentation fault (core dumped)
```

randaz81, 2015-03-13:
It crashes if a single part (i.e. left_arm) is requested but not available

lornat75, 2015-03-16:
Commit 463de0f9e8377980659c45c8dfd0a9d8b0188dd1 checks for yarpserver 
Commit 2df7c82fb22ddba1dca678441726f44e0e691e1a fixes crash when a part is not available

There are still two issues:

1) log messages do not show up in the console, so errors during startup are not printed. I haven't fixed it because this happens for all the LOG and LOG_ERROR calls in main.cpp

2) The gui closes when a single part is unavailable, without attempting to open the other ones. This requires probably more structured changes.

aitek4iit, 2015-03-17:
For the first point I can not figure out what is the problem. The LOG and LOG_ERROR are macro for fprintf. If it is compiled in debug it will log using qDebug else using fprintf. 

For the second point I'll take a look nad try to solve it

aitek4iit, 2015-03-17:
Can I test this with iCub_SIM? Can I disable a part? In this case, how can I do?

barbalberto, 2015-03-17:
the file share/yarp/contexts/simConfig/iCub_parts_activation.ini has a tag [PARTS] with the list of active parts, you can disable one by setting it to 'off'

aitek4iit, 2015-03-17:
@barbalberto thanks!
I've just pushed a fix for the second point

pattacini, 2015-03-17:
I can see warning messages popping up when a part is unavailable.
Yet, no message is displayed when `yarpserver` is not found: minor issue, any how.

aitek4iit, 2015-03-17:
Ok. Fixed and pushed

commit 0a3bcb86 (2015-03-17): fixed #440 thanks @aitek4iit
closed 2015-03-17 by pattacini