robotology/yarp #441: yarpmotorgui-qt does not close ports on closure

alecive, 2015-03-12:
When closing the yarpmotorgui qt, the ports are not closed and are stil available from `yarp name list`.

```
alecive@malakim:~/Programmazione/iCub/icub-main/build (master)$ yarpmotorgui --robot icubSim
||| configuring
||| no policy found
||| default config file specified as yarpmotorgui.ini
||| checking [/home/alecive/Programmazione/iCub/icub-main/build/yarpmotorgui.ini] (pwd)
||| checking [/home/alecive/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/alecive/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| found /home/alecive/.local/share/yarp/robots/default
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/ICUBcontrib/robots/default] (robot yarp.d)
||| checking [/usr/local/share/iCub/robots/default] (robot yarp.d)
||| checking [/home/alecive/.local/share/yarp/robots/default/yarpmotorgui.ini] (robot)
||| checking [/home/alecive/.config/yarp/yarpmotorgui.ini] (YARP_CONFIG_HOME)
||| checking [/home/alecive/.local/share/yarp/yarpmotorgui.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/ICUBcontrib/yarpmotorgui.ini] (yarp.d)
||| checking [/usr/local/share/iCub/yarpmotorgui.ini] (yarp.d)
||| did not find yarpmotorgui.ini
yarp: cannot read from yarpmotorgui.ini
icub
Setting default parts.
APPENDING  "head" 
APPENDING  "torso" 
APPENDING  "left_arm" 
APPENDING  "right_arm" 
APPENDING  "left_leg" 
APPENDING  "right_leg" 
Checking the existence of: /icubSim/yarpmotorgui0/head/rpc:o 
ADDRESS is:  
yarp: Port /icubSim/yarpmotorgui0/head/rpc:o active at tcp://10.255.10.167:10044
yarp: Port /icubSim/yarpmotorgui0/head/command:o active at tcp://10.255.10.167:10045
yarp: Port /icubSim/yarpmotorgui0/head/state:i active at tcp://10.255.10.167:10046
yarp: Port /icubSim/yarpmotorgui0/head/stateExt:i active at tcp://10.255.10.167:10047
yarp: Sending output from /icubSim/yarpmotorgui0/head/rpc:o to /icubSim/head/rpc:i using tcp
yarp: Sending output from /icubSim/yarpmotorgui0/head/command:o to /icubSim/head/command:i using udp
yarp: Receiving input from /icubSim/head/state:o to /icubSim/yarpmotorgui0/head/state:i using udp
yarp: Receiving input from /icubSim/head/stateExt:o to /icubSim/yarpmotorgui0/head/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
Setting a valid finder 
yarp: Port /yarpmotorgui/head/sequence:o active at tcp://10.255.10.167:10048
Opening interfaces...Allocating memory 
head iencs->getEncoders() ok!
Checking the existence of: /icubSim/yarpmotorgui0/torso/rpc:o 
ADDRESS is:  
yarp: Port /icubSim/yarpmotorgui0/torso/rpc:o active at tcp://10.255.10.167:10049
yarp: Port /icubSim/yarpmotorgui0/torso/command:o active at tcp://10.255.10.167:10050
yarp: Port /icubSim/yarpmotorgui0/torso/state:i active at tcp://10.255.10.167:10051
yarp: Port /icubSim/yarpmotorgui0/torso/stateExt:i active at tcp://10.255.10.167:10052
yarp: Sending output from /icubSim/yarpmotorgui0/torso/rpc:o to /icubSim/torso/rpc:i using tcp
yarp: Sending output from /icubSim/yarpmotorgui0/torso/command:o to /icubSim/torso/command:i using udp
yarp: Receiving input from /icubSim/torso/state:o to /icubSim/yarpmotorgui0/torso/state:i using udp
[... 72 lines cut ...]
registration name /icubSim/skin/torso_comp ip 10.255.10.167 port 10018 type tcp
registration name /icubSim/skinManager/skin_events:o ip 10.255.10.167 port 10013 type tcp
registration name /icubSim/texture/face ip 10.255.10.167 port 10042 type tcp
registration name /icubSim/torso/command:i ip 10.255.10.167 port 10039 type tcp
registration name /icubSim/torso/rpc:i ip 10.255.10.167 port 10038 type tcp
registration name /icubSim/torso/state:o ip 10.255.10.167 port 10040 type tcp
registration name /icubSim/torso/stateExt:o ip 10.255.10.167 port 10041 type tcp
registration name /icubSim/world ip 10.255.10.167 port 10002 type tcp
registration name /icubSim/yarpmotorgui0/head/command:o ip 10.255.10.167 port 10045 type tcp
registration name /icubSim/yarpmotorgui0/head/rpc:o ip 10.255.10.167 port 10044 type tcp
registration name /icubSim/yarpmotorgui0/head/state:i ip 10.255.10.167 port 10046 type tcp
registration name /icubSim/yarpmotorgui0/head/stateExt:i ip 10.255.10.167 port 10047 type tcp
registration name /icubSim/yarpmotorgui0/left_arm/command:o ip 10.255.10.167 port 10055 type tcp
registration name /icubSim/yarpmotorgui0/left_arm/rpc:o ip 10.255.10.167 port 10054 type tcp
registration name /icubSim/yarpmotorgui0/left_arm/state:i ip 10.255.10.167 port 10056 type tcp
registration name /icubSim/yarpmotorgui0/left_arm/stateExt:i ip 10.255.10.167 port 10057 type tcp
registration name /icubSim/yarpmotorgui0/right_arm/command:o ip 10.255.10.167 port 10060 type tcp
registration name /icubSim/yarpmotorgui0/right_arm/rpc:o ip 10.255.10.167 port 10059 type tcp
registration name /icubSim/yarpmotorgui0/right_arm/state:i ip 10.255.10.167 port 10061 type tcp
registration name /icubSim/yarpmotorgui0/right_arm/stateExt:i ip 10.255.10.167 port 10062 type tcp
registration name /icubSim/yarpmotorgui0/torso/command:o ip 10.255.10.167 port 10050 type tcp
registration name /icubSim/yarpmotorgui0/torso/rpc:o ip 10.255.10.167 port 10049 type tcp
registration name /icubSim/yarpmotorgui0/torso/state:i ip 10.255.10.167 port 10051 type tcp
registration name /icubSim/yarpmotorgui0/torso/stateExt:i ip 10.255.10.167 port 10052 type tcp
```

aitek4iit, 2015-03-16:
I've committed a fix. Please try it

pattacini, 2015-03-16:
It works on Windows.
@alecive please verify on linux too.

alecive, 2015-03-16:
@pattacini It works!
@aitek4iit is it possible to close the gui and then close the ports? It would make more sense to have this behavior rather than the gui closing after few seconds because it's waiting for the ports to get closed :+1:

aitek4iit, 2015-03-17:
The only thing I can do is to hide the window when the close event is rceived. I can not close the gui because the ports are closed in the distructors of the single parts widgets.
If the hiding the main window could solve your problem then I'll do it

aitek4iit, 2015-03-17:
I've just pushed a fix

alecive, 2015-03-18:
@aitek4iit the fix works, but it does not apply when you are issuing `Ctrl+C` on the terminal window. Can you fix that as well?

alecive, 2015-03-18:
@aitek4iit also, when closing the gui by clicking on the close button, I get an error while freeing memory:
```
yarp: Removing output from /icubSim/yarpmotorgui0/torso/rpc:o to /icubSim/torso/rpc:i
yarp: Removing output from /icubSim/yarpmotorgui0/torso/command:o to /icubSim/torso/command:i
yarp: Removing input from /icubSim/torso/state:o to /icubSim/yarpmotorgui0/torso/state:i
yarp: Removing input from /icubSim/torso/stateExt:o to /icubSim/yarpmotorgui0/torso/stateExt:i
*** Error in `yarpmotorgui': free(): invalid pointer: 0x000000000226d020 ***
```

aitek4iit, 2015-03-19:
Ok, I'll take a look

aitek4iit, 2015-03-23:
I've pushed a fix that should solve it.

lornat75, 2015-04-16:
@alecive, can we close this issue?

alecive, 2015-04-17:
I still have this problem. And it is actually worse now:
```
yarp: Removing output from /icubSim/yarpmotorgui0/torso/rpc:o to /icubSim/torso/rpc:i
yarp: Removing output from /icubSim/yarpmotorgui0/torso/command:o to /icubSim/torso/command:i
yarp: Removing input from /icubSim/torso/state:o to /icubSim/yarpmotorgui0/torso/state:i
yarp: Removing input from /icubSim/torso/stateExt:o to /icubSim/yarpmotorgui0/torso/stateExt:i
*** Error in `yarpmotorgui': free(): invalid pointer: 0x000000000183aef0 ***
Aborted (core dumped)
```
It ends up with an abort.

aitek4iit, 2015-04-17:
Can you tell me what did you exactly do? I can not reprdoduce this behavior. I'm trying on windows and on ubuntu.

alecive, 2015-04-17:
Yes sure:
 * terminal 1 : `yarpserver --write`
 * terminal 2 : `iCub_SIM`
 * terminal 3: `yarpmotorgui` - this is the ini file that I'm using (for disabling the legs automatically):
```
alecive@malakim:~/Programmazione/yarp/build (master)$ cat /home/alecive/.local/share/yarp/robots/default/yarpmotorgui.ini
//name of the robot          
name icubSim      
//parts to be opened by the GUI            
parts (head torso left_arm right_arm)
 
[head_zero]    
//                          0             1             2             3             4             5    
PositionZero                0             0             0             0             0             0    
VelocityZero                10            10            10            10            10            10   

[torso_zero]   
//                          0             1             2              
PositionZero                0             0             0              
VelocityZero                10            10            10             
 
[left_arm_zero]              
//Joint                     0             1             2             3             4             5             6             7             8             9             10            11            12            13            14            15             
PositionZero                -30           30            0             45            0             0             0             15            30            0             0             0             0             0             0             0              
VelocityZero                10            10            10            10            30            30            30            100           100           100           100           100           100           100           100           100            
 
[right_arm_zero]             
//Joint                     0             1             2             3             4             5             6             7             8             9             10            11            12            13            14            15             
PositionZero                -30           30            0             45            0             0             0             15            30            0             0             0             0             0             0             0              
VelocityZero                10            10            10            10            30            30            30            100           100           100           100           100           100           100           100           100            
 
[right_leg_zero]             
//Joint                     0             1             2             3             4             5              
PositionZero                0.00          8.00          0.00          0.00          0.00          -8.00          
VelocityZero                10.00         10.00         10.00         10.00         10.00         10.00          
 
[left_leg_zero]              
//Joint                     0             1             2             3             4             5              
PositionZero                0             8             0             0             0             -8             
VelocityZero                10            10            10            10            10            10             
 
[head_calib]   
//                          0             1             2             3             4             5              
CalibrationType             3             3             3             3             0             0              
Calibration1                2047.5        2047.5        2047.5        2047.5        900           900            
Calibration2                10            10            10            10            20            20             
Calibration3                1870.05       3293.063      404.95        558.5125      0             0         

//DO NOT REMOVE THIS LINE   
```

The, wait for the `yarpmotorgui` to open, and after closure it aborts. For the sake of completeness, I closed it with the "x" button on the window.

aitek4iit, 2015-04-17:
Ok, this reproduce the bug. I found that crash something while i try to delete a Bottle:

the bottle is:
Bottle *pParts=finder->find("parts").asList();

and when I delete it using

delete pParts;

it crashes.

pattacini, 2015-04-17:
This is an expected behavior since you don't have the ownership to delete that object. In other words, the call to the method `asList()` gives back a pointer to access internal data.

Everything will be properly disposed as soon as the `finder` object will be deleted.

aitek4iit, 2015-04-17:
Ok. Fixed. I never noticved this because I did not tryied it using yarpmanager.ini.

closed 2015-05-05 by alecive