robotology/yarp #442: [BufferedPort] Lock when setting callback and not opening the port

francesco-romano, 2015-03-13:
If a callback is set (`useCallback()`) on a port which for some reason will not be opened then the `close()` instead of doing nothing remained lock on a semaphore.

It seems that the `useCallback` method changes the internal state of the port: indeed calling `isClosed()` on the port before setting the callback returns true, while calling it after returns false.

francesco-romano, 2015-03-13:
I attach a sample project: https://dl.dropboxusercontent.com/u/7968745/YarpReadTest.zip

cc @drdanz

drdanz, 2015-03-13:
Commenting out `attachIfNeeded()` in `useCallback` seems to fix the example, but I don't know if this is the correct fix or if it will cause any other problem later...

```diff
diff --git a/src/libYARP_OS/include/yarp/os/BufferedPort.h b/src/libYARP_OS/include/yarp/os/BufferedPort.h
index e4ff135..bc73fa2 100644
--- a/src/libYARP_OS/include/yarp/os/BufferedPort.h
+++ b/src/libYARP_OS/include/yarp/os/BufferedPort.h
@@ -274,7 +274,7 @@ public:
      * available.
      */
     void useCallback(TypedReaderCallback<T>& callback) {
-        attachIfNeeded();
+        //attachIfNeeded();
         reader.useCallback(callback);
     }
 
```

drdanz, 2015-03-13:
Another way to fix it seems to be to call `port->disableCallback();` before the `port->close();` line, I'm not sure whether it should be before or after `port->interrupt();` (or if it is indifferent).

```diff
--- a/src/Class.h
+++ b/src/Class.h
@@ -35,6 +35,7 @@ virtual ~ClassModule()
                 std::cerr << "NO\n";
             }
             port->interrupt();
+            port->disableCallback();
             port->close();
 
             delete port;
```

drdanz, 2015-03-13:
The difference in the second case is that the output writes "closed? NO" instead of "closed? YES"

drdanz, 2015-03-13:
I added some unit tests that show this behaviour, for now they are commented out, we should enable them when they are working.

paulfitz, 2015-03-13:
I think an alternative way to fix this would be, in PortReaderBuffer.h, before any calls to `delete reader`, to do `reader->stop()`. This will make sure that if there is a reader thread started, it is shut down. This could also be done from the reader's destructor.

drdanz, 2015-03-13:
Fixed in db5016f43c59cce33d14f8e7272ed51a59e16eb7
Thanks @paulfitz

closed 2015-03-13 by drdanz

francesco-romano, 2015-03-13:
@drdanz @paulfitz 
This commit fixed the lock while closing the port.

Still, if the port has not been opened but a callback has been associated with it, the `isClosed()` method return false.