robotology/yarp #443: I select a control mode for a joint in yarpmotorgui-qt but I get another one

randaz81, 2015-03-13:
when I select a control mode (e.g. torque) on a joint (e.g. joint 5 of the leg), sometimes the joint switches to a different control mode (e.g. velocity). This is most likely a bug of the gui.
Additionally, there's probably a bug  (TBV) in yarp controlboard wrapper about the velocityMove single joint method, which sends the command to different joint.

randaz81, 2015-03-13:
I confirm a bug in yarp for the second point. It can be reproduced even without using the yarpmotorgui-qt. Just run "simpleClient --robot icub --part left_leg" and then send the command "set vmo 5 0.0". The command will be received by joint 0 instead of 5. Tested on ETH robot (it merges two devices: upper and lower leg). @barbalberto

barbalberto, 2015-03-13:
Fixed!

randaz81, 2015-03-13:
@barbalberto perfect! thank you
@aitek4iit the original issue about the control mode changes in the gui menus is still present.

aitek4iit, 2015-03-23:
I saw that behaviour when I came in iit, but it seems to be a control board problem. Do you still have this problem? I can not reproduce it using the simulator.

barbalberto, 2015-03-23:
I just pulled yarp & iCub and the bug happens to me with yarpmotorgui-qt and the iCub_SIM. When I set torque I get velocity mode instead. It is ok with the robotMotorGui

aitek4iit, 2015-03-23:
Ok. I think it is now fixed!

alecive, 2015-03-23:
@aitek4iit this is the error I'm having:
```
[ 79%] Building CXX object src/yarpmotorgui-qt/CMakeFiles/yarpmotorgui-qt.dir/jointitem.cpp.o
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/jointitem.cpp: In constructor ‘JointItem::JointItem(int, QWidget*)’:
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/jointitem.cpp:159:49: error: ‘JointState’ is not a class or namespace
     ui->comboMode->setItemData( IDLE,           JointState::Idle, Qt::UserRole);
                                                 ^
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/jointitem.cpp:160:49: error: ‘JointState’ is not a class or namespace
     ui->comboMode->setItemData( POSITION,       JointState::Position, Qt::UserRole );
                                                 ^
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/jointitem.cpp:161:49: error: ‘JointState’ is not a class or namespace
     ui->comboMode->setItemData( POSITION_DIR,   JointState::PositionDirect, Qt::UserRole );
                                                 ^
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/jointitem.cpp:162:49: error: ‘JointState’ is not a class or namespace
     ui->comboMode->setItemData( MIXED,          JointState::Mixed, Qt::UserRole );
                                                 ^
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/jointitem.cpp:163:49: error: ‘JointState’ is not a class or namespace
     ui->comboMode->setItemData( VELOCITY,       JointState::Velocity, Qt::UserRole );
                                                 ^
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/jointitem.cpp:164:49: error: ‘JointState’ is not a class or namespace
     ui->comboMode->setItemData( TORQUE,         JointState::Torque, Qt::UserRole );
                                                 ^
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/jointitem.cpp:165:49: error: ‘JointState’ is not a class or namespace
     ui->comboMode->setItemData( OPENLOOP,       JointState::OpenLoop, Qt::UserRole );
                                                 ^
make[2]: *** [src/yarpmotorgui-qt/CMakeFiles/yarpmotorgui-qt.dir/jointitem.cpp.o] Error 1
make[1]: *** [src/yarpmotorgui-qt/CMakeFiles/yarpmotorgui-qt.dir/all] Error 2
make: *** [all] Error 2
```

aitek4iit, 2015-03-24:
Strange. On windows platform I don't have this error. However i should have fixed this bug just now.

gsaponaro, 2015-03-24:
I confirm that with bc55eb99f028dcbfd7bd4c6770cc96b521d7d533 Linux compilation works.

randaz81, 2015-03-29:
issue "unable to switch control mode when current mode is velocity"  now fixed 
  43e6d977fea2f9f83442a6bbc6ace37605e2ea9a

closed 2015-03-29 by randaz81

randaz81, 2015-03-29:
similar issue was affecting buttons 'run', 'home', etc. when control mode == velocity
now fixed in: bd6ffdc8c5269d5c3c6cbd8be462b1753f5a977f