robotology/yarp #445: includes in thrift

tanismar, 2015-03-17:
Hi, I am trying to use a thrift structure definition from the objects3D library (within the grasp application) in another application I am building so that the modules from my application can communicate using this structure directly via thrift. 
However, although thrift allows includes, they can only be found "using the current path, or by
 * searching relative to any paths specified with the -I compiler flag. " citing the thrift tutorial. How can we set the CMakeLists, or the include, so that libraries added there are also available for the thrift compilation?

Here is my thrift code, which if i comment the showCloud line, compiles fine, but not otherwise:
#tool3Dshow.thrift

include "objects3D.thrift"

/**
* tool3DShow_IDLServer Interface.
*/

service tool3Dshow_IDLServer
{
    // setVerbose (ON/OFF) - sets verbose of the output on or off
    // @return true/false on success/failure of setting verbose
    bool setName(1: string name);

    // show - Opens the visualizer and displays the PointCloud from file
    // @return true/false on showing the poitncloud
    bool showFromFile(1: string verb);

    // show - Opens the visualizer and displays the received  PointCloud
    // @return true/false on showing the poitncloud
    bool showCloud(1: objects3D.SurfaceMeshWithBoundingBox mesh);

    // setVerbose (ON/OFF) - sets verbose of the output on or off
    // @return true/false on success/failure of setting verbose
    bool setVerbose(1: string verb);

    // help - produces help with longer descritpion of each command and its parameters.
    // @return true/false on success/failure of extracting features
    bool help_commands();
}

elen4, 2015-03-31:
Hi Tanis, AFAIK currently there is no mechanism in place to have inter-package dependent Thrift files. The `yarp_idl_to_dir` macro currently calls the Thrift generator with `-I <directory_from_where_the macro_is_called>`. I guess that the macro could be expanded to "include" user-defined directories. Ideally, packages that contain Thrift files (in this case, the objects3D library) should also install them and export their location through CMake, so that you could easily provide the location to the macro. This should allow you to generate code for your file, then to compile and use this code you should also remember to include header files for the objects3D library, ad link against it. This is just the general idea but I haven't really thought it through. As a quick, inelegant and not recommended workaround, if you want to test whether your Thrift file works, you could just copy objects3D.thrift in the same directory. Hope this helps a bit...

tanismar, 2015-03-31:
Hi Elena, 
Indeed, thrift allows in principle to include .h files, but as you say it seems the YARP implementation does not. I actually opted for the inelegant solution, but it means having the same libraries copied all over the place where you need them, instead of just referenced. although for the moment is ok. 
I hope the macro gets improved at some point to allow for normal includes, that would considerably expand the usability of thrift, specially for its use to serialize user defined data types. 
Thanks in any case!

closed 2015-11-27 by tanismar