robotology/yarp #446: [yarpmotorgui] Unused variable warnings

drdanz, 2015-03-18:
They are quite annoying, I don't know whether it's safe to get rid of these variables or if they are unused because something is missing.

@aitek4iit can you have a look at them and get rid of the warnings in the qt version, please?
Someone else who knows the gtk version can take care of it? @randaz81 perhaps?

**yarpmotorgui-gtk**

```
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/guiPid2.cpp: In function ‘void guiPid2::change_page(GtkNotebook*, GtkWidget*, guint, gpointer)’:
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/guiPid2.cpp:73:14: warning: variable ‘color_green’ set but not used [-Wunused-but-set-variable]
     GdkColor color_green;
              ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/guiPid2.cpp:74:14: warning: variable ‘color_red’ set but not used [-Wunused-but-set-variable]
     GdkColor color_red;
              ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/guiPid2.cpp:75:14: warning: variable ‘color_pink’ set but not used [-Wunused-but-set-variable]
     GdkColor color_pink;
              ^
```

```
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp: In function ‘void myMain2(GtkButton*, int*)’:
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:164:26: warning: unused variable ‘note4’ [-Wunused-variable]
                     gint note4 = gtk_notebook_append_page((GtkNotebook*) nb1, main_vbox4, label);
                          ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:343:26: warning: unused variable ‘note5’ [-Wunused-variable]
                     gint note5 = gtk_notebook_append_page((GtkNotebook*) nb1, main_vbox5, label);
                          ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:120:16: warning: unused variable ‘main_vbox1’ [-Wunused-variable]
     GtkWidget *main_vbox1        = NULL;
                ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:121:16: warning: unused variable ‘main_vbox2’ [-Wunused-variable]
     GtkWidget *main_vbox2        = NULL;
                ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:122:16: warning: unused variable ‘main_vbox3’ [-Wunused-variable]
     GtkWidget *main_vbox3        = NULL;
                ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:143:10: warning: unused variable ‘legsLabel’ [-Wunused-variable]
     char legsLabel[]= "Legs";
          ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp: In function ‘GtkTreeModel* create_net_model()’:
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:505:10: warning: unused variable ‘i’ [-Wunused-variable]
     gint i = 0;
          ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp: At global scope:
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:503:23: warning: ‘GtkTreeModel* create_net_model()’ defined but not used [-Wunused-function]
 static GtkTreeModel * create_net_model (void)
                       ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/main.cpp:526:13: warning: ‘void combo_net_changed(GtkComboBox*, gpointer)’ defined but not used [-Wunused-function]
 static void combo_net_changed (GtkComboBox *box,    gpointer   user_data)
             ^
```

```
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/multipleJointWindow.cpp: In static member function ‘static void partMover::fixed_time_move(const double*, double, partMover*)’:
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/multipleJointWindow.cpp:37:22: warning: unused variable ‘iamp’ [-Wunused-variable]
   IAmplifierControl *iamp = currentPart->amp;
                      ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/multipleJointWindow.cpp:38:16: warning: unused variable ‘ipid’ [-Wunused-variable]
   IPidControl *ipid = currentPart->pid;
                ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/multipleJointWindow.cpp:39:19: warning: unused variable ‘itrq’ [-Wunused-variable]
   ITorqueControl *itrq= currentPart->trq;
                   ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/multipleJointWindow.cpp:41:8: warning: unused variable ‘SEQUENCE_TMP’ [-Wunused-variable]
   int *SEQUENCE_TMP = currentPart->SEQUENCE;
        ^
/opt/iit/src/yarp/src/yarpmotorgui-gtk/src/multipleJointWindow.cpp:42:11: warning: unused variable ‘TIMING_TMP’ [-Wunused-variable]
   double *TIMING_TMP = currentPart->TIMING;
           ^
[... 772 lines cut ...]
         ^
```

```
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp: In constructor ‘PartItem::PartItem(QString, QString, yarp::os::ResourceFinder*, bool, bool, bool, bool, bool, QWidget*)’:
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp:168:10: warning: unused variable ‘ok2’ [-Wunused-variable]
     bool ok2 = true;
          ^
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp: In member function ‘void PartItem::onCalibClicked(JointItem*)’:
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp:540:15: warning: unused variable ‘jointIndex’ [-Wunused-variable]
     const int jointIndex = joint->getJointIndex();
               ^
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp: In member function ‘void PartItem::onPidClicked(JointItem*)’:
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp:571:12: warning: variable ‘debug_param’ set but not used [-Wunused-but-set-variable]
     double debug_param [8];
            ^
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp: In member function ‘void PartItem::onOpenSequence()’:
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp:1079:9: warning: variable ‘totPositions’ set but not used [-Wunused-but-set-variable]
     int totPositions = 0;
         ^
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp: In member function ‘void PartItem::updatePart()’:
/opt/iit/src/yarp/src/yarpmotorgui-qt/partitem.cpp:1678:10: warning: unused variable ‘buffer’ [-Wunused-variable]
     char buffer[40] = {'i', 'n', 'i', 't'};
          ^
```

lornat75, 2015-03-18:
Looks like these warnings originate from our code, we can fix them ourselves!

drdanz, 2015-03-18:
sure, whoever wrote that part ;)

alecive, 2015-03-18:
When I compile the motorgui, I have this error (that instead should gui-related):

```
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/mainwindow.cpp:106:9: warning: unused variable ‘w’ [-Wunused-variable]
     int w = metrics.width(globalLabel);
         ^
```

I think that it's safe to remove the line. @aitek4iit what do you think?

lornat75, 2015-03-19:
Fixed some warnings: #451

aitek4iit, 2015-03-24:
I removed the warnings belonging to the gui right now.

lornat75, 2015-03-24:
can we close this issue?

alecive, 2015-03-24:
I still have these:
```
[ 96%] Building CXX object src/yarpmotorgui-qt/CMakeFiles/yarpmotorgui-qt.dir/mainwindow.cpp.o
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/main.cpp: In function ‘int main(int, char**)’:
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/main.cpp:114:10: warning: variable ‘deleteParts’ set but not used [-Wunused-but-set-variable]
     bool deleteParts=false;
          ^
[ 96%] Building CXX object src/yarpmotorgui-qt/CMakeFiles/yarpmotorgui-qt.dir/partitem.cpp.o
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/partitem.cpp: In member function ‘void PartItem::onPidClicked(JointItem*)’:
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/partitem.cpp:574:12: warning: variable ‘debug_param’ set but not used [-Wunused-but-set-variable]
     double debug_param [8];
            ^
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/partitem.cpp: In member function ‘void PartItem::onOpenSequence()’:
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/partitem.cpp:1082:9: warning: variable ‘totPositions’ set but not used [-Wunused-but-set-variable]
     int totPositions = 0;
         ^
In file included from /home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/partitem.cpp:10:0:
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/partitem.cpp: In member function ‘void PartItem::onSaveSequence(QList<SequenceItem>)’:
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/log.h:22:50: warning: format not a string literal and no format arguments [-Wformat-security]
 #define LOG_ERROR(...) fprintf(stderr,__VA_ARGS__)
                                                  ^
/home/alecive/Programmazione/yarp/src/yarpmotorgui-qt/partitem.cpp:1178:5: note: in expansion of macro ‘LOG_ERROR’
     LOG_ERROR(QString("Saving File %1\n").arg(completeFileName).toLatin1().data());

```

drdanz, 2015-03-25:
@lornat75 all the warnings from the gtk version are still there, so I wouldn't close this issue yet

pattacini, 2015-03-25:
@drdanz I wouldn't maintain gtk version any longer...

lornat75, 2015-03-25:
Sorry but I consider this a minor issue.

drdanz, 2015-03-25:
Me too, that's why it was labelled with `Severity: Minor` :stuck_out_tongue_closed_eyes: 
Anyway I suppressed the warnings for the gtk version, but there are still warnings in the qt version (the ones reported by @alecive), so I'm reopening this.

pattacini, 2015-09-02:
very minor issue

closed 2015-09-02 by pattacini