robotology/yarp #452: [YARP to ROS] Sending string to a ROS topic adds a "\x00" to the end of the string

mikearagao, 2015-03-20:
I have code on Github that reproduces this behavior but maybe you can test it with some code of yours.

I'm adding the string to the bottle like this:

    bottleToSend.add("the string");

To run it I'm doing the following:

    // terminal 1
    $ roscore
    // terminal 2
    $ yarpserver --ros
    // terminal 3
    $ yarpidl_rosmsg --web true --name /typ@/yarpidl
    // terminal 4
    $ roslaunch my_package my_launch
    // terminal 5
    $ ./my_yarp_code
    // terminal 6
    $ rostopic echo /the_topic
        field: "the string\x00"

I'll gladly provide the instructions to run my code in order to test this. Actually the repositories containing already have almost all the instructions needed.

Cheers!

paulfitz, 2015-03-26:
Thanks for reporting @mikearagao. Yes, could you give instructions for replicating this? With a quick test like this ...
```c++
#include <yarp/os/all.h>
using namespace yarp::os;
int main() {
  Network yarp;
  Port port;
  port.promiseType(Type::byNameOnWire("std_msgs/String"));
  port.open("/my/topic+@/my/node");
  int ct = 0;
  while (true) {
    Bottle msg;
    msg.addString(ConstString("message #") + Bottle::toString(ct));
    printf("Writing: %s\n", msg.get(0).asString().c_str());
    port.write(msg);
    Time::delay(1);
    ct++;
  }
  return 0;
}
```
... everything looks in order:
```
$ rostopic echo /my/topic 
data: message #14
---
data: message #15
---
data: message #16
---
...
```
No doubt there's a problem but I'd appreciate your help narrowing it down.

closed 2015-04-01 by paulfitz

mikearagao, 2015-04-01:
Great! @lornat75 and @paulfitz just tested this with my master thesis' module and everything is working now!

commit 569b7d70 (2015-04-01): Merge pull request #461 from paulfitz/string_tweak

[ros] omit final null character when sending strings, fixes #452
commit 974784c5 (2015-04-01): [ros] omit final null character when sending strings, fixes #452

For compatibility with an ancient version of yarp, strings have been transmitted
with a null character included. Non-ancient versions of yarp will accept either
a string with or without a null termination. This commit stops transmitting that
null termination, but preserves the ability to receive a null terminated string
for compatibility between programs using different versions of yarp.