robotology/yarp #453: [YARP to ROS] Sending empty list causes unwanted error

mikearagao, 2015-03-20:
I'm trying to send an empty list to ROS and I get the following message:

    [ERROR]Structure of message is unexpected (expected sensor_msgs/JointState)
    [ERROR]Expected 'list of float', got 'list of mixed'

ROS allows us to send an empty list, so I tried to add an empty list to my bottle like this:

    Bottle& list_5 = message.addList();

If I add something to the list it works:

    list_5.add(0.0);
    list_5.add(0.0);

But I want to send an empty list on some specific examples...
In order to run this I am performing the same instructions that I've written on issue #452 

Cheers :smile:

mikearagao, 2015-03-23:
Hi @paulfitz, do you think this is enough? As I said I can provide code to test this... (I've created both this and issue #452)

paulfitz, 2015-03-26:
thanks @mikearagao, was able to replicate with this:
```c++
#include <yarp/os/all.h>
using namespace yarp::os;
int main() {
  Network yarp;
  Port port;
  port.promiseType(Type::byNameOnWire("sensor_msgs/JointState"));
  port.open("/my/joints+@/my/node");
  int ct = 0;
  while (true) {
    Bottle msg;
    // header
    Bottle& hdr = msg.addList();
    hdr.addInt(ct);
    Bottle& time = hdr.addList();
    time.addInt(0);
    time.addInt(0);
    hdr.addString("0");
    // payload
    msg.addList().addString("test");  // name[]
    msg.addList().addDouble(1);  // position[]
    msg.addList().addDouble(1);  // velocity[]
    msg.addList()/*.addDouble(1)*/;  // effort[]
    printf("Writing: %s\n", msg.toString().c_str());
    port.write(msg);
    Time::delay(1);
    ct++;
  }
  return 0;
}
```
I think I'm also now seeing #452 in the `name` field. Will work on fixes.

mikearagao, 2015-03-26:
Got it! I'll wait for the fixes!

Actually I'm adding the string to the bottle with the `add("test")` instead of the `addString("test")`, but since you're also seeing the issue #452 I guess that's not the reason.

commit 8e305514 (2015-03-27): [ros] send empty lists correctly when translating on fly, fixes #453
closed 2015-03-27 by paulfitz