robotology/yarp #457: [Qt - OS X] Menubar does not work until app unfocused and focused again

francesco-romano, 2015-03-25:
This issue happens on any Qt5 app on OS X (10.10 for sure. I don't know in previous versions).
Also it seems it is not linked to a particular qt version. I have Qt 5.4.1, but it has been noticed also on 5.3.2.

When the app is launched the menubar is not updated (see picture)
[image]

In order to properly update the menubar I have to unfocus the application (e.g. click on the desktop) and then click back on the app.
Any click on the "wrong" menubar does not lead to any action. The bar simply does not respond.

@vtikha

vtikha, 2015-03-25:
Yes, also noted on Qt 5.3 and higher. It seems it only works by switching applications and returning to the actual GUI. Maybe its related to [this](http://doc.qt.digia.com/qt-5.2/macosx-issues.html)

kt10aan, 2015-03-25:
See this issue #340 
The patch posted there works for yarpmanager, but for the other apps it needs a bit more tweaking. I meant to do this and submit the patches but I haven't  got a glimpse of free time yet.

vtikha, 2015-09-02:
Came back to this issue and as @kt10aan mentioned in previous issue #340 (but simplifying it extremely) I have added  `MACOSX_BUNDLE ` in all qt application to build the executable as a proper application bundle on Mac OS X. This makes it a GUI executable that can be launched from Spotlight with a working menubar. If the .app itself is added to the application folder then it will also work from the terminal (but I would avoid doing this) Spotlight is good enough :smile:

francesco-romano, 2015-09-02:
There is an issue if the apps are installed and rpath is used.
The library path is not added to the `RPATH` variable, and so the yarp libraries are not loaded.

A fix is here: https://github.com/robotology/yarp/commit/f4a11830d6b031e30665521a67f277c4ceab8a00

@vtikha  can you check that the apps still work on your configuration?
Also another system volunteer will be appreciated  ( @pattacini @traversaro)

@drdanz I don't know if this is the best cmake code :)

@vtikha  I still have issues with yarpscope and yarpview:

~~~
Makas-Pro:bin makaveli$ ./yarpview.app/Contents/MacOS/yarpview 
QQmlApplicationEngine failed to load component
qrc:/qml/QtYARPView/main.qml:20 "qrc:/YARPView/": no such directory

Makas-Pro:bin makaveli$ ./yarpscope.app/Contents/MacOS/yarpscope 
QQmlApplicationEngine failed to load component
qrc:/qml/QtYARPScope/main.qml:21 "qrc:/YARPScope/": no such directory
~~~

vtikha, 2015-09-02:
@francesco-romano The fix is ok on my configuration and qt apps such as `yarpmanager`, `yarpdataplayer` `yarplogger` `yarpmotorgui` work perfectly with spotlight. I also confirm that `yarpview` and `yarpscope` do not open with spotlight but they do using the terminal. I therefore reopen this issue until all apps are working correctly on OS X.

francesco-romano, 2015-09-05:
@vtikha  we were not too far yesterday..
Qt searches the following paths to look for the plugins:
~~~
QQmlImportDatabase::addImportPath: "/usr/local/Cellar/qt5/5.4.2/qml"
QQmlImportDatabase::addImportPath: "/Users/makaveli/Projects/src/local/bin/yarpview.app/Contents/MacOS"
QQmlImportDatabase::addImportPath: "/Users/makaveli/Projects/src/local/bin/yarpview.app/Contents/lib/qt5/qml"
QQmlImportDatabase::addImportPath: "/Users/makaveli/Projects/src/local/bin/yarpview.app/lib/qt5/qml/Release"
~~~

The least ugly path is (imho): `/Users/makaveli/Projects/src/local/bin/yarpview.app/Contents/lib/qt5/qml`, so adding (manually) the files installed in `lib/qt5/qml` solved the issue.
Now the problem is to properly configure the cmake. 
If I understood correctly the code we should "mimic" these lines
```
install(TARGETS QtYARPViewPlugin
        COMPONENT utilities
        RUNTIME DESTINATION "${CMAKE_INSTALL_QMLDIR}/${QtYARPViewPlugin_QMLDIR}"
        LIBRARY DESTINATION "${CMAKE_INSTALL_QMLDIR}/${QtYARPViewPlugin_QMLDIR}"
        ARCHIVE DESTINATION "${CMAKE_INSTALL_QMLDIR}/${QtYARPViewPlugin_QMLDIR}")

install(FILES qmldir DESTINATION "${CMAKE_INSTALL_QMLDIR}/${QtYARPViewPlugin_QMLDIR}")
```
but instead of installing them we have to copy them into the app bundle.
Here I have to problems:

1. this information resides in another CMake (the plugin one) and not the app.
2. I don't know how to set the `set_source_files_properties(${APP_ICON} PROPERTIES MACOSX_PACKAGE_LOCATION "")` for a target

Maybe here @drdanz can help us

vtikha, 2015-09-06:
:+1: Indeed we were not that far! Let's figure out the best way to do this with @drdanz & @lornat75

drdanz, 2016-06-20:
@francesco-romano Is this still an issue?

francesco-romano, 2016-06-20:
@claudiofantacci in all the QT applications?

claudiofantacci, 2016-06-20:
It seems so. Are there any particular apps you want me to try?

vtikha, 2016-06-20:
@francesco-romano I do not think that it has changed much from the previous Qt version used above. I confirm that it is fine (using the same setup as @claudiofantacci) with ```yarpdataplayer``` & ```yarpmotorgui``` but still have issues with ```yarpview``` & ```yarpscope```.

francesco-romano, 2016-06-20:
Ok. So nobody solved the `yarpview` and `yarpscope`. Now I remember the issue.

Then, @drdanz this issue is not solved.

vtikha, 2016-06-20:
@claudiofantacci can you please provide the output of ```otool -L yarpview``` for example? please run this in the ```$YARP_DIR/bin``` folder

claudiofantacci, 2016-06-20:
I have no problem with any of those program...! Everything seems to work fine to me!

Here it is the output of ```otool -L yarpview```
yarpview:
	/usr/local/opt/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.6.0, current version 5.6.1)
	/usr/local/opt/qt5/lib/QtQuick.framework/Versions/5/QtQuick (compatibility version 5.6.0, current version 5.6.1)
	/usr/local/opt/qt5/lib/QtQml.framework/Versions/5/QtQml (compatibility version 5.6.0, current version 5.6.1)
	/usr/local/opt/qt5/lib/QtNetwork.framework/Versions/5/QtNetwork (compatibility version 5.6.0, current version 5.6.1)
	/usr/local/opt/qt5/lib/QtGui.framework/Versions/5/QtGui (compatibility version 5.6.0, current version 5.6.1)
	/usr/local/opt/qt5/lib/QtCore.framework/Versions/5/QtCore (compatibility version 5.6.0, current version 5.6.1)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)

francesco-romano, 2016-06-20:
@claudiofantacci are you using the command line version or the `.app` version?

claudiofantacci, 2016-06-20:
I tried with both command line and .app.

The only problem I have is that the .app version of ```yarpview``` do not even start.

vtikha, 2016-06-20:
@claudiofantacci ok, for the app this is to be expected as it is missing some files in order to load properly. ```yarpview``` and ```yarpscope``` have a different way of loading files. But I still have the issues with the command line menu being unresponsive until app is unfocused and focused again....

claudiofantacci, 2016-06-20:
Do you have my system settings and experiencing the issue? This is weird...!

commit 844949bc (2016-06-21): Fixed macOS .app bundle for YARPView and YARPScope

Embedded in .app bundle plugins' .so and qmldir file

Fixes #457
PR #806 (2016-06-21): Fixed macOS .app bundle for YARPView and YARPScope
Embedded in .app bundle plugins' .so and qmldir file

Fixes #457 

Thanks @drdanz
<a href='#crh-start'></a><a href='#crh-data-%7B%22processed%22%3A%20%5B%22https%3A//github.com/robotology/yarp/pull/806%23issuecomment-227195156%22%5D%2C%20%22comments%22%3A%20%7B%22General%20Comment%22%3A%20%7B%22html_url%22%3A%20%22https%3A//github.com/robotology/yarp/pull/806%23issuecomment-227195156%22%2C%20%22comments%22%3A%20%5B%7B%22body%22%3A%20%22cc%20%40claudiofantacci%20%20%40vtikha%20%22%2C%20%22created_at%22%3A%20%222016-06-20T16%3A30%3A54Z%22%2C%20% [... cut]
closed 2016-06-21 by drdanz