robotology/yarp #458: Problem with yarpmotorgui sliders when playing with arrow keys

pattacini, 2015-03-26:
In `position mode`, when I select the position slider and start off pressing the arrow keys on the keyboard I'm not able to modify the set-point. Further, I can modify the velocity slider beneath with the arrow keys but the velocity values does not change accordingly.

/cc @alecive @randaz81 @drdanz

randaz81, 2015-03-29:
Similar issues present also in other control modes:
in `velocity mode` i am able to move the slider with the arrow keys, but the number above the slider does not change and the velocity command is not executed.
Other modes (torque, mixed etc) behave like `position mode`: pressing the arrow keys initially moves the slider, but then the command is overwritten by the current value.

aitek4iit, 2015-04-17:
Fixed

pattacini, 2015-04-17:
The problem of controlling sliders by means of arrow keys is sorted out: great.

However, the fact that the slider is set to track the current encoder in `position mode` prevents from using the arrow keys effectively. In the gtk gui, the slider was always on the desired target, that is it was not following the feedback.

/cc @randaz81 @alecive

alecive, 2015-05-05:
@pattacini the issue should be fixed now. Please check it out.

closed 2015-05-05 by alecive

pattacini, 2015-05-05:
it works! thanks.