robotology/yarp #459: Bug with "[include file.ini]  when parsing .ini config files

barbalberto, 2015-03-27:
When using keyword [include file.ini] sometimes part of the data already stored are lost.
The result depends on where the keyword [include ] is placed along the file:
if all [include file.ini] directives are at the beginning of the file it works, if some data are already present in the .ini file and then we add an [include file.ini], previous data are lost.

Example:
This works
[include "include_one.ini"]
[include "include_two.ini"]

[root_group]
ciao hello
bau miao

[end_group]
lili  lolo
cici  bubu
-------------------------------
while this does not work:
[include "include_one.ini"]

[root_group]
ciao hello
bau miao

[include "include_two.ini"]

[end_group]
lili  lolo
cici  bubu

In the second case, if I read it with 
    yarp::os::Property prop.fromConfigFile(file.ini) 
the whole group [root_group] does not appear in the final property object, while [end_group] is ok.

I did a small test case here:
https://gist.github.com/7b45d991e92869b53691.git

you can run it with ./buil/test_include  (pwd should be the one containing the .ini files)

@lornat75 any idea?

traversaro, 2015-09-02:
The gist for reproducing the issue is https://gist.github.com/barbalberto/7b45d991e92869b53691 .

traversaro, 2015-09-02:
Bug confirmed in regression test added in https://github.com/traversaro/yarp/tree/fixIssue459 .

traversaro, 2015-09-02:
Bug solved in https://github.com/robotology/yarp/issues/571 .

closed 2015-09-02 by traversaro