robotology/yarp #460: Appearance of yarpmotorgui-qt on linux systems is bad

randaz81, 2015-03-27:
on windows yarpmotorgui-qt is ok, but on linux several widgets and labels are truncated
[image]

alecive, 2015-03-27:
@randaz81 as a matter of fact, this happens on the debian machines on the clusters, but it does not happen on Ubuntu.

randaz81, 2015-04-09:
@alecive tested on Ubuntu 14.04, the problem is the same.
@aitek4iit can you check this issue?

aitek4iit, 2015-04-10:
I have to a set up an ubuntu environment. As soon as I can i'll check it. Howevere I think it's matter of fonts: I don't specify any fints, so each system uses it's own default fonts. I think that specifynig fonts and font size could solve the issue.

lornat75, 2015-04-10:
if setting up an ubuntu system takes time consider coming up to visit us!

drdanz, 2015-04-10:
That's very likely to be a problem with the qt5 style on debian and on some ubuntu... @alecive which Ubuntu version are you using?

aitek4iit, 2015-04-17:
I have just committed a fix that could solve the problem. Can you check it please?

alecive, 2015-04-17:
It is working for me. But we have to test it on the cluster. @aitek4iit what did you do? Did you specify a font?

alecive, 2015-04-17:
Btw, I remember that this was also happening with other gui, such as `frameGrabberGui2`, but I have to investigate further.

aitek4iit, 2015-04-17:
Yes, I specified a font and a font size: Actually is "Tahoma 8pt", that is the same as on windows. Linux uses a font bigger than windows by default.

alecive, 2015-04-17:
Ok. Tahoma 8pt is not installed on linux systems (it is a windows font), so you might encounter more problems over the time because it fallbacks to some other fonts. But on my machine everything works (probably because it still uses the `8pt` directive), so it's ok. I'll keep you updated for the debian machines as soon as I get one.

lornat75, 2015-04-17:
Maybe it falls back to a different font, but it keeps the size, this is why it works.

Anyway: is it possible to specify different fonts on windows and linux? in this case we could select a font that is available on both systems.

aitek4iit, 2015-04-17:
Yes it is possible using #ifdefs

lornat75, 2015-04-22:
@alecive any chance you tested it?

alecive, 2015-04-27:
I did (sorry if I wasn't clear). As I said in my previous comment, on my machine behaves well, but it might be that on some other machines this is not the case. We can neither foresee nor test any possible user setting. Therefore, I suggest to use a more standard font, such as `Arial` or whatever suitable for the task (arial is not great for quite a number of reason, but it would fix the issue).

aitek4iit, 2015-04-27:
I think we can specify only the point size of the font and not the family. This could solve the problem, and would let the software use the system default font.

alecive, 2015-04-27:
@aitek4iit if that is the case, let's go with it :)

aitek4iit, 2015-04-27:
Ok. Fixed. Can you give a try?

alecive, 2015-04-27:
It works, thanks! Closing.
@aitek4iit for the future: it would be useful if you reference the issue you're fixing into your github commit message. In this way, we can better track your commits in the issues directly! Take a look at this https://guides.github.com/features/issues/#notifications

closed 2015-04-27 by alecive